prusa3d/PrusaSlicer #7681: Sending gcode to Octoprint if .3mf is on network crashes PrusaSlicer V2.4

btCRO, 2022-01-04:
### Version
Prusa MK3S+

PrusaSlicer-2.4.0+win64-202112211614

### Operating system type + version
Windows 10 (build 19044), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Prusa MK3S+ with MMU2S

### Behavior
After slicing any file that is located on network drive if you try to send it to printer (OctoPrint) application crashes with error:
"Unhandled unknown exeption; terminating application" and after that "Internal error: invalid string position". Saving gcode works fine, problem is only when you try to send it to printer.

When I copy sam 3MF to local hard drive, there is no error and it is sliced file is sent to printer just fine.

Previous versions worked fine, problem started after installing V2.4. Tried reinstalling application, no difference.

bubnikv, 2022-01-04:
Would you please provide the 3MF for review?

So you slice a STL at the network drive, slice and send to OctoPrint? Then it crashes?
Would you please share the complete path of the STL you loaded? Are there any spaces or non-ASCII characters in the path?

btCRO, 2022-01-08:
I wasn't able to add 3MF here, it states that file type is not supported.

Yes, it only crashes if file is opened from network path, if I copy same file to desktop everything works as it should.
I cannot see connection between gcode that needs to be sended to Octoprint and path file resides in 3MF format.
It crashes only if you try to send file to Octoprint (after presing button) and if you try to save it localy everything's fine.

I tried original file that is not ASCII and then ASCII and there is no difference.

I tried to install fresh install on new PC and file gets sent to Octoprint once and after that behaviour is the same as on my computer, it crashes with same error. 

Then I tried to run Wizzard and remove user profile printer, even on my computer slicing and sending to Octoprint works once, and on next file it crashes again.

bubnikv, 2022-01-08:
Thanks. You are the only one reporting such an issue. We tested your
scenario as we understand it. We wonder whether there is something special
about your network path. How do you enter the network path? With
```
\\yourserver\path\to\your\stl\model.stl
```
?

Is there anything special about that server? About the path? Any localized
characters? For example Russian, Czech, Arabic whatever?

You said that after a fresh install opening an stl from a network file
system, slicing and sending to OctoPrint worked, but at the 2nd and further
tries slicer crashes?

You may attach 3MF to github if you zip it first. We contacted github to
allow for 3mf files, but we did not receive any response.

bubnikv, 2022-01-08:
> It crashes only if you try to send file to Octoprint (after presing
button) and if you try to save it localy everything's fine.

Does it crash when you click the "Send button" at the plater? Or first
after you click "Send" at the OctoPrint dialog where you may modify the
final G-code file name?

bubnikv, 2022-01-08:
The error report indicates that some string (text in computer jargon) is
accessed outside of its bounds. If it crashed, we would ask you to enable
collection of crash dumps, however this one throws an exception (again
computer jargon), thus it crashes much later at a spot which has nothing to
do with the original trigger.

What language is PrusaSlicer configured to communicate with you? The issue
may be dependent on translation dictionary.

btCRO, 2022-01-08:
this is network path for one of the tried parts:
"P:\TEHNIČKI ODJEL\Roboti\Elementi za kordinatne sustave\Stožac 80mm.3mf"

Path includes Croatian alpabet, I think that you allso have those letters in your local alphabet.
I tried to copy file on desktop in new folder, if folder includes non ASCII alphabet, it crashed same as with network path, so problem is using folder with non ASCII letters (file has them and in file name there is no issue detected). 
When folder is renamed so it contains only ASCII, Slicer did not crashed.

btCRO, 2022-01-08:
I use it in English language.
Can you tell me how to enable collection of crash dumps so I can send it?

btCRO, 2022-01-08:
And regarding when it crashes,

It crashes after presing G button, but **does not** offer popup where I can change name of file before sending gcode to printer.

So it happens right pressing button.

bubnikv, 2022-01-08:
Thanks. We are getting close, I believe our testers will be able to
reproduce on Monday. Once we are able to reproduce it, we will fix it
easily.

My bet is that the last used path is processed somewhere and the local
characters are mangled.

kocikdav, 2022-01-10:
Hello @btCRO would you please be so kind and try this [build](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/DK/PrusaSlicer-2.4.0-beta2+43-dk-printhost-win64-ge398289d3-202112010937.zip)?
If the download requires login use login: prusaslicer password: slicer.

Please run the build as prusa-slicer-console.exe. It should still crash same way as before, but it should log into console exactly where the crash is happening. Please copy the full log here. Thanks!

btCRO, 2022-01-11:
Hello,

I've downloaded that version and it seems that program does not crashes.
I've noticed even that in this version non-ASCII characters are replaced with ASCII variant before sending to printer.

In release version2.4.0+win64-202112211614 if file have non-ASCII character it names that file with odd letters (does not read them well), but on this beta version (Š is replaced vith S) and nothing crashes :)

kocikdav, 2022-01-11:
That is interesting. We do have algorithms that changes letters into ASCII ones but those should have worked before as well. 

Does the path you are uploading to contains any non-ASCII characters now?

btCRO, 2022-01-12:
Yes, it is interesting. Beta version works as it shoud, but release version does not change letters. Path and name of the document was the same.
Today I tried again on my original network path, release version of PrusaSlicer crashes before opening popup with ability to change name before sending it to printer but this beta version did everything as it should and did not crash using same file on same path.

kocikdav, 2022-01-14:
@btCRO I apologize. I've mismatched links to build I have send to you. Correct link is:
https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/DK/PrusaSlicer-2.4.0+33-dk-upload-debug-win64-gc41c1f169-202201101042.zip 
Would you please try this one for us? 
Your previous testing is still valuable as we can pinpoint better where we made the error. So thanks :)

kocikdav, 2022-01-17:
I added possible fix, but the log of crash from version I posted above would still help greatly.

btCRO, 2022-01-17:
Hello,

Well, I ‘we downloaded this new version, started console and again, this versions swaps letters with accent like ŠĐČĆŽ into SDCCZ and everything works as it should. I have problems only with production version so far. Both versions you sent to try work just fine.
I’ve added pictures, you can see both how they display name of same file. Hope that helps a bit.

Test version:
[image]

And Production version:
[image]

btCRO, 2022-01-18:
Hi,
I'we tested this version with same files, as I can tell you for now, I cannot see any problems, all files are sliced, letters swaped, slicer works and send all files to printer.

Thank you.

closed 2022-01-18 by kocikdav

SirFlour, 2022-01-26:
hope that helps, i am not sure. I have a same (version 2.4.0+win64) issue. When just opening a .3mf file on my pc and not saving it after it loaded, it will crash trying to upload the sliced file into Octoprint. So i can solve it whit just saving in Prusa Slicer and then it can be uploaded. 

will that be also fixed with the new version? Or should i open a new issue?

ErrorMsg: "Unhandled unknown exception; terminating the application" - interner Fehler: invalid string position 09:20:51: Interner Fehler: invalid string position
09:20:51: Interner Fehler: invalid string position

kocikdav, 2022-01-26:
@SirFlour Thanks, thats a useful observation. Can I ask you to try the same with the build I uploaded above? [Or here.](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/DK/PrusaSlicer-2.4.1-alpha0+18-win64-g2cc20594a-202201171440.zip)

SirFlour, 2022-01-26:
Hello kocikdav, yes it works in the alpha :) many thanks for the fast responce and i am looling forward to the new version :)

So long it helps to save before slicing and uploading :)

bubnikv, 2022-01-28:
@SirFlour 

>  When just opening a .3mf file on my pc and not saving it after it loaded, it will crash trying to upload the sliced file into Octoprint. 

What 3MF did you load and how? Would you please share the 3MF? What is your system language and PrusaSlicer language? German? Are there any German characters (umlauts, sharp s) in the path to the 3MF file?

It may be, that some of the German characters get mangled on import.

SirFlour, 2022-01-29:
@bubnikv yes, that it is in my case. i use german in both setups. i saved it onto my desktop - C:\Users\xxx xxüx\Desktop and have the ü in the path. I open it and save it, then it works. the 3MF name looks okay. so i guess prusaslicer solves the ü while saving. also it works with the alpha version very well, without any problems :) many thanks your great support. if i can help / test anything more, just let me know :)

bubnikv, 2022-02-15:
Anybody is able to find a way to get the mangled characters into the application bar with PrusaSlicer 2.4.1-beta1?