prusa3d/PrusaSlicer #7682: ability to add layer height to an added part in the slicer

Lietzen, 2022-01-04:
### Version
prusa slicer 2.4.0

Version 2.4.0+win64

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Prusa MK3s+

### Behavior

I just loaded an object, and then added a part to my object, but i wanted that part to be printed more detailed, because its a hinge, and i want it to be as smooth as possible. but to do that i need to split the object, and then the part lands on the bed, and is no longer located at the correct height, so i neede to split the object, and then relocate it back to the position.

But prusaslicer dosnt allow free floading parts so i need to add a generic-box. so i can move the object up.

i should just be allowed to select the part i wanted to be printed in another layer height. but the option isnt there..

Please add the featue to select a part of a print, and modify the layer height of that part. :-) 

[image]

[image]

bubnikv, 2022-01-04:
[image]

[image]

[image]

[image]

Snuff1eupagus, 2022-01-04:
The OP is correct. The method described has been the only way to do this for some time now, in order to have different layer heights you must use separate objects. Parts in an assembly can not currently have their own layer height. 

The OP is not referring to "Height range modifiers" as this would modify all the added Parts in a given assembly at the specified height range. (An assembly is what an Object becomes after you add a part to it) the original object itself becomes part of the assembly and only the assemblies and individual objects allow for separate layer height settings.
 

An Object or Assembly can have these settings.
[image]

A part added to an object and the original object itself (which becomes a part in the assembly) only have access to these settings.
[image]

While having the ability to add different layer heights to individual parts of an assembly would be nice, if we could place our objects in 3D space not just XY without having to add a part (to maintain contract with the bed) we could then have a better work flow and gain more control over our prints without the hassle of a work around, and the befuddled repositioning that follows, this would allow us to import separate objects from our 3D design software, have them maintain their position in 3d space and have all the benefits of being able to set the properties for the individual objects, it would also be desirable to be able to import an assembly of parts and "split to objects" having them maintain their coordinate space. 

This would of course be for expert users.

Regards

n8bot, 2022-01-04:
Thank you for the clarifications and correction. I misunderstood both the problem, and the way that parts work vs objects. 

Let me see if there is some other workaround, and/or how difficult of a change it would be to allow parts with different layer heights in the same object. 

I fear it might be very difficult. In the code, there is bot much distinction between a part and an object. There is mostly just print, then object, then region. I suppose that the parts are actually regions. I'll have to look. A chance for me to learn! (Reminder that I'm not part of the prusa team just a tinkerer.)

Snuff1eupagus, 2022-01-05:
@n8bot  I'm aware your not, nor am I, but I know things ; ) and I've noticed you know things too.

I'm happy there is at least a workaround. The difficulties are why I suggested the possible alternative, making it possible to have "Objects" maintain a position of their own (in the Z) above the build surface, this would give the same capability in effect (objects can already have there own layer heights) now they just need to be able to hold a position above the bed. This would reduce the number of steps needed to achieve the goal.

Just a thought,

Thanks for looking

n8bot, 2022-01-05:
Wait, what workaround? It seems like there is no way for a print-in-place part to have a different layer height independent o the parent object/other parts in the object.

Lietzen, 2022-01-06:
The work around is like so.

1. Add your stl file
2. Add part to your print file (and write down the Z position of that part)
3. split the parts to objects (The part drops to the bed)
4. Now right click the part that dropped to the bed, and select add a new part, and select box
5. Now the part can be mooved, up, but to get it to the same position we need to do a little math.
- Find the z position of the generic box (my case -45.16) and the Z position of the part (my case 49,3)
- Add the two values together and subtract the generic box heigt which is 12.5
- So the math looks like this (45,16+49,3-12,5) That will bring the part back to its position
6. Now to make sure the generic box isnt printed, reduce the infill, layers and perimiters to 0
7. Now you can select the part, and choose another layer heigt for that part.

This is unneccessary complicated, because clearly prusa slicer can handle printing in different layer height for different objects, but the feature just simply isnt available, under layers and perimiters, and it should be. or at least give us the option to split parts into objects, keepin, the original position.

Hope this all makes sense :-)

n8bot, 2022-01-06:
Omfg!!! Lol. I'm sorry. That's quite the workaround! A better solution needs to be made.

Snuff1eupagus, 2022-01-07:
Whats sad is I actually started doing some form of this in Slic3r, the workaround is actually easier now then it was then, and still the developers don't seem to comprehend the need. I'm not talking about the ability to add different layer heights, I referring to the ability to have an object maintain a position in 3D space in Slicer in the Z axis.

n8bot, 2022-01-07:
Yes, I understand exactly what you mean. I agree it should be easier to force objects or parts to arbitrary heights.

closed 2026-09-29 by prusaslicer-bot[bot]