prusa3d/PrusaSlicer #7686: pothole on next perimeter after retract

gitpirojoke, 2022-01-04:
### Version
2.4.0

### Operating system type + version
win7-64

### 3D printer brand / version + firmware version (if known)
Flying Bear Ghost 5

### Behavior
if the retract was on the perimeter in front of the outer one, a pit is formed on the outer perimeter

#### Project File (.3MF) where problem occurs
[top.zip](https://github.com/prusa3d/PrusaSlicer/files/7810257/top.zip)
[image]
[image]
[image]
[image]

n8bot, 2022-01-05:
I think this problem can be improved with some of the changes talked about here: https://github.com/prusa3d/PrusaSlicer/pull/7483#issuecomment-992708731

I think the next version should have improved seams. The prusa team is on it!

rtyr, 2022-01-06:
I printed your g-code as it is on the Prusa MINI. I can see only slight bulge caused by internal seam.

Your g-code.
[image]

bubnikv, 2022-01-28:
The seam placement issues will be improved with the upcoming PrusaSlicer 2.4.1-beta1 release.
In particular, we reverted staggering of the inner seams wrt. to the outer seams, which was introduced in 2.4.0.

closed 2022-01-28 by bubnikv

n8bot, 2022-01-28:
I liked the staggered seam placement. I suppose you found through testing that it had no actual benefit?

I'll have to do a test print to compare. I was getting as close to perfect seams as I could imagine while using 2.4.0.