prusa3d/PrusaSlicer #7687: Raft overlaps Prusa Slicer 2.4

juckendesAuge, 2022-01-04:
### Version
Version 2.4.0+linux-x64
sha256sum PrusaSlicer-2.4.0+linux-x64-202112211614.AppImage 
b6c34157ff2acffae5b39a1997f5694e9ca4717d5c9b370f75a6728eeadb9cab  PrusaSlicer-2.4.0+linux-x64-202112211614.AppImage

### Operating system type + version
_What OS are you using, and state any version #s_
cat /etc/manjaro-release 
Manjaro Linux
uname -a
Linux SelmaT400 5.15.12-1-MANJARO #1 SMP PREEMPT Wed Dec 29 18:08:07 UTC 2021 x86_64 GNU/Linux

_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
FLSUN Q5 with Firmware Version 2.5. Stock with 0.6mm Nozzle

### Behavior
* _Describe the problem_
Hello,
In the raft overlap the raft layers. I have multiplied a hook 6 times. The 6 hooks are printed all at once. I have set 3 raft layers. Each raft layer overlaps.
This can lead to a clogged nozzle. Also, the layers are not flat when overlapped.

* _Steps needed to reproduce the problem_
go to print settings/support material/raft layers and insert 3
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
no

[image]
[image]
[image]
[image]
[image]

[3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7811068/3mf.zip)

rtyr, 2022-01-05:
Duplicate of #4517, #2501, #4788. As of now, there is no collision detection of individual objects, it is up to the user to prevent them.

Closing as a duplicate.

closed 2022-01-05 by rtyr

viper0078, 2022-01-05:
#7573 has the solution.