prusa3d/PrusaSlicer #7688: Missing Pieces of Text and Loss of Detail When Slicing in Version 2.4

sigaby60, 2022-01-05:
### Version
Version 2.4.0+win64

### Operating system type + version
**Copied from OS specifications:**
Edition	Windows 11 Home
Version	21H2
Installed on	‎11/‎8/‎2021
OS build	22000.376
Experience	Windows Feature Experience Pack 1000.22000.376.0

**Copied from PrusaSlicer:**
Version:   2.4.0+win64
Build:     PrusaSlicer-2.4.0+win64-202112211614

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 22000), 64-bit edition
Total RAM size [MB]: 34,253MB
OpenGL installation
GL version:   4.6.0 NVIDIA 466.77
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 3070/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
[... 358 lines cut ...]
</details>

### 3D printer brand / version + firmware version (if known)
Prusa i3 MKS+ stock with no modifications
Firmware 3.10.0-4481

### Behavior
* _Summary of the problem_
I imported a STL file into PrusaSlicer 2.4. The 3D editor view looked exactly like the original file I created in Fusion360. After slicing the model, the preview showed portions of the text missing as well as finer details. Any adjustments to the print settings did not produce different results.
* _Steps needed to reproduce the problem_
  1. Import the STL file into PrusaSlicer 2.4
  2. Slice the model
  3. The preview now shows disconnected portions of the text on the model, a missing colon and degree symbols.
* _Expected Results_
I expected the preview to look exactly like the 3D editor view after slicing the model.
* _Actual Results_
After slicing the model, the preview did not look like the model in the 3D editor view. Portions of the text on the model were disconnected and the colon symbol is missing in one area as well as the degree symbols. However, slicing the model in 2.3.3 worked as expected.

#### Screenshots of the 3D Editor View and Preview:
[image]
[image]
_Is this a new feature request?_
No.
#### Project File (.3MF) where problem occurs
[Filament Sample.zip](https://github.com/prusa3d/PrusaSlicer/files/7811710/Filament.Sample.zip)

locki-cz, 2022-01-05:
Problem is on your side, use nozzle 0,25 or different font which is wide, because your fornt have very thin walls. When you enable thin walls detection it show something more.

sigaby60, 2022-01-05:
> Problem is on your side, use nozzle 0,25 or different font which is wide, because your fornt have very thin walls. When you enable thin walls detection it show something more.

When I enable thin wall detection, yes it shows a little more but r and t letters are suddenly missing portions. However, in version 2.3.3 I don't have any of these problems whatsoever
[image]
.

Area5142, 2022-01-06:
In PS version 2.4.0, another solution is to use _Print Settings -> Advanced -> XY Size Compensation_ to make the text larger and easier to print. A size of 0.1-0.2mm would be my suggestion.

sigaby60, 2022-01-07:
> In PS version 2.4.0, another solution is to use _Print Settings -> Advanced -> XY Size Compensation_ to make the text larger and easier to print. A size of 0.1-0.2mm would be my suggestion.

This solution worked great and got me the result I was looking for. Thank you.

closed 2022-01-07 by sigaby60