prusa3d/PrusaSlicer #7692: Prusa oneri

Saltundal, 2022-01-05:
Merhaba,

Fuzzy skin güzel bir ozellik. Daha kullanışlı olmalı. istediğim yüzeye fuzzy skin yapmak için , yüzey seçme ozelliği eklemelisiniz.
İlk katman yüksekliği ayarlananıyor ama , ilk katmanı 100% yapınca etek (brim) değeri ile aynı. Etek (brim) ve ilk katman ayrı ayrı olmalı.

bubnikv, 2022-01-05:
Why don't you use `https://translate.google.com/` or some other free translate service? We are sorry, but we don't understand Turkish.

> Fuzzy skin is a nice feature. It should be more useful. In order to make a fuzzy skin on the surface I want, you must add a surface selection feature.
The first layer height is adjusted, but when you make the first layer 100%, it is the same as the skirt (brim) value. Skirt (brim) and first layer should be separate.

bubnikv, 2022-01-05:
And the automatic translate is not understandable.

Saltundal, 2022-01-05:
Hi, Fuzzy skin is a nice feature. It should be more useful. Add a surface select feature to make a blurry complexion. The first layer straightens the line but is the same as the hem value when first adjusted. Skirt and first layer should be separate.

closed 2026-09-29 by prusaslicer-bot[bot]