prusa3d/PrusaSlicer #7699: Avoid crossing perimeters creates scars on outer perimeters

Moyrn, 2022-01-06:
### Version
2.4.0

### Operating system type + version
Win10-64

### 3D printer brand / version + firmware version (if known)
MK3S MMU2S

### Behavior
ticking the box on "avoid crossing perimeters" now induces some "moves" on the outer perimeter which end in creating vertical scars.
disable it, and the scar disables. 
It was present in beta 2 as well (just tested), but was OK in 2.3.x versions and also OK in 2.4.0 alpha1 (although that one had issues with paint scars+rear locations -- see photo)

* _Describe the problem_
i'm printing with silk pla. very sensitive to surface artifacts. So i disable retractation, print items one by one, and use "avoid crossing perimeters". All is fine since years with Prusa Slicer.
Now i'm trying to print again the same object, and i saw vertical scars on the sides - and this is not the seam-scar (paint seam + rear are properly managed).

* _Steps needed to reproduce the problem_
tick "'avoid perimeters", slice. display "moves", and see the scar one 2 sides.
untick, and the scars disappears (but then I have stringing inside my object because i have disabled retractation)

  * _If this is a command-line slicing issue, include the options used_
  no

* _Expected Results_
avoid crossing "external" perimeter as previously

* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
2.4.0
[image]

same project in 2.4.0 alpha 1 : no scars on the two sides (but we see the seam scar does not follow paint+rear but this is already fixed in later versions)
[image]

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
[BDIS12-B.zip](https://github.com/prusa3d/PrusaSlicer/files/7824310/BDIS12-B.zip)

Many thanks for your very kind help, and my apologies if I miss something.

Take care you all

W

Moyrn, 2022-01-06:
seems ok in alpha 3 as well (with seam scar fixed)

n8bot, 2022-01-06:
I see what you are talking about. It seems caused by the travel moves coming too close to, and actually exiting, the part boundary. They for sure don't seem to account for the diameter of the nozzle... this could be it.

[image][image]

Moyrn, 2022-01-07:
Hello

Yes. Interesting !!
Just sharing thoughts, i also force the ext perimeter width to 0.3 (nozzle
0.45) to get detail prints on the letters.
But when i force b ack to 0.4 or 0.45, the move-scar still appears on the
preview.

Maybe in the computation should the travel take into account exact
ext-perim-width instead of nozzle
or simply avoid ext perimeters - i believe - when there are inner
perimeters of course.

In any way, i confirm it was still ok in Alpha3

thanks for your devoted support !

Wilfried

Moyrn, 2022-01-07:
and fine comment on the external/total nozzle width !!

Snuff1eupagus, 2022-01-07:
@Moyrn

I've found trying to get 0.3 extrusion width from a 0.4 nozzle to be troublesome, I tend to not go lower then 0.35 max with a 0.4 nozzle, and if i need to go smaller it's time to switch nozzles. In the .3mf example you posted if you adjust your external perimeters and perimeters to 0.35 each the problem appears to go away, and the detail maintained.

[image]

[image]

It might be worth taking a look,

Regards

n8bot, 2022-01-07:
While it looks like the problem goes away, look at hoe closely the blue travel lines are still crossing the external perimeter. That line is where the center of the nozzle is. The nozzle is likely 1.0 mm in diameter. Do the math! Any nozzle-print contact, especially if the nozzle is hot, will definitely leave a mark. Z-hop could definitely alleviate the issue, but retraction is disabled so I forget if one can have z hop with 0 retraction, or if it would just lead to a stringy mess anyways.

IMO, the travel moves should try to tuck themselves into the model better.

Moyrn, 2022-01-08:
Hello

Thanks for the comment to help me "avoid" the issue. Much appreciated.
I know, sometimes switching nozzle is the (only) answer. But it try to
avoid changing when I can (i already killed a MK3S head because of wires,
got leaking, etc... + i get a better compromise speed/quality tweaking
around with 0.4). But I do not neglect the advice. Do not worry.

But on the other hand
- no problem with cura - even in 0.25
- no problem with prusa slicer in 2.4.0 alpha 3 (and earlier, 2.1, 2.2,
2.3...)

So to avoid the problem, i simply reinstalled alpha3 - which prints very
fine.

I think as n8bot says, travel should take into account physical size of
nozzle & the ext perim width.
Which seems not impossible - again - as it was ok before that.

Moyrn, 2022-01-17:
Hello

I moved my mini to nozzle 0.25 to get better results, but still sometimes need to go to alpha 3 for scar alignment issue
(https://github.com/prusa3d/PrusaSlicer/issues/7720)

But i however still face this perimeter-crossing moves on my MK3S + nozzle 0.40 with perimeters 0.40.
So that may indicate the issue is not linked to my low perim width settings. Agains this is OK in alpha 3
So i design the project in 2.4.0 and issue the gcode from alpha3.

Here are pics. 2.4.0
[image]

2.4.0 Alpha 3
[image]

@PrusaSlicer team: thanks for your attention & help...

lukasmatena, 2022-01-17:
This should now be fixed by @lukashejl, the fix will be included in 2.4.1 patch release. Closing.

closed 2022-01-17 by lukasmatena

Moyrn, 2022-01-17:
Much appreciated. Thanks.