prusa3d/PrusaSlicer #7701: Auto deployment uninstall line

brandsADM, 2022-01-07:
### Version
2.4.0

Hi there,

we want to deploy via Intune PrusaSlicer to a set of school PCs. Installation works fine, but the uninstall command ends with error.
The command "C:\Program Files\Prusa3D\unins000.exe" /sp- /silent /norestart" is surely not correct.

Any idea  about the right command?

Thanks.

msnapka, 2022-01-07:
As I implemented it today in our school, I found out the following results:

PrusaSlicer silent installation:
prusa3d_win_2_4_0.exe /VERYSILENT /SUPPRESSMSGBOXES

PrusaSlicer installs two instead of a single app (I mean, wtf mates...):
1) Prusa3D (seems like just bloatware, links, etc.)
2) PrusaSlicer (the app we actually need)

Whole pack uninstall:
1) uninstall PrusaSlicer first
2) uninstall Prusa3D second

...otherwise you will spot some data leftovers in start menu, registry, program files, etc.

PrusaSlicer silent uninstall:
"C:\Program Files\Prusa3D\PrusaSlicer\unins000.exe" /SILENT
Prusa3D silent uninstall:
"C:\Program Files\Prusa3D\unins000.exe" /SP- /VERYSILENT

More docs on the installer:
https://jrsoftware.org/ishelp/index.php?topic=setupcmdline

More docs on many more installers:
http://unattended.sourceforge.net/installers.php

...or open up admin cmd and type blahblah.exe /?

Usually it throws out something like the PrusaSlicer:
[image]

All the best,

Marek Šnapka
IT admin
https://www.zemedelka-opava.cz

Scush22, 2022-01-12:
Hi msnapka,

I am also trying to get Prusa to do a silent install and yet still getting install windows and boxes appearing using the exact same exe and arguments as yourself

I was wondering if there might be something I am missing in order to silently install successfully?

lukasmatena, 2022-01-12:
@Janys

msnapka, 2022-01-12:
> Hi msnapka,
> 
> I am also trying to get Prusa to do a silent install and yet still getting install windows and boxes appearing using the exact same exe and arguments as yourself
> 
> I was wondering if there might be something I am missing in order to silently install successfully?

You are not doing it wrong, the silent installation is not respected. However I am installing the product using a service account and remote installation tools (PDQ Deploy). This runs in a separate user context, so it does not throw out anything even it wanted to. Prusa Slicer authors definitely have to work on this part. I tried to report it a few times, however nobody answers the phone in the company and nobody actually replied to my e-mail after a month, which is rather unpleasant. So maybe somebody will notice it at least here.

brandsADM, 2022-01-21:
Uninstall is working fine now.

The hint from @msnapka to handle Prusa like two single packages is the right way. Two Intune app-packages with the different uninstall commands above remove Prusa3d completly.

Thanks for your help.

jake-the-creative, 2022-01-25:
The lack of an actually silent install isn't great, it's hampering my rollout.

Scush22, 2022-01-26:
The workaround I've gone with is to install slicer on a machine and then rebundle the installation back into my own installer that respects silent flags

It's a bit of a pain to sort out but it works just fine for deployment

msnapka, 2022-03-29:
> The workaround I've gone with is to install slicer on a machine and then rebundle the installation back into my own installer that respects silent flags
> 
> It's a bit of a pain to sort out but it works just fine for deployment

Scush22 - could You please elaborate on repackaging the program?

@lukasmatena You tagged somebody - if you know anyone from the dev team or directly are a part of the dev team, could you, please, push on prioritization of this issue? Thanks.

Scush22, 2022-03-29:
So I used a program called NSIS which allows you to repackage files into an installer, the support is good enough that I figured out how to use it after playing around with it for about an hour

Basically you install Prusa using the official installer to get the files and then create a folder somewhere that contains the nsis bundle script you write and another folder that contains the installation files then when you run the script it will package everything up and allow you to install Prusa as normal but it will listen to silent flags

The actual process of writing the script is probably too in depth for me to do in a way that makes sense and I am unsure on the ethics of sharing the installer but I can share privately the script which should allow you figure the rest out

msnapka, 2022-03-29:
Thank You. I have today sent an email to Prusa to prioritize this thread. I suppose that anyone interested in finally solving the software installation automation will get in touch with You if necessary. Thanks again.

Scush22, 2022-03-29:
> Thank You. I have today sent an email to Prusa to prioritize this thread. I suppose that anyone interested in finally solving the software installation automation will get in touch with You if necessary. Thanks again.

Apparently private messaging is not an option so for now I will just post here and hopefully once the developers fix this on the official version we can all use that instead

Ok so the steps to use this
- Install NSIS
- Install Prusa
- Download the install script from **[here]**(https://www.mediafire.com/file/5raka7j4syrzxy6/PrusaSlicerInstall.nsi/file)
- Create a new folder (I created one in documents)
- In this folder create a new folder called PrusaSlicer and copy all files from the official install into this folder
- Copy the PrusaSlicerInstall.nsi into the root of the folder you created
- Compile the script with NSIS to create your installer
- The created installer can be installed silently and installs just as any normal program would

I hope this helps anyone that needs it

StewLanco, 2022-08-01:
Lack of silent install support creates issues for our company. It's very hard to keep everyone on the same version. This is one of the few applications we still have to manually install and manage. Please dedicate some development resources to solving this.

closed 2024-02-29 by lukasmatena