prusa3d/PrusaSlicer #7702: PrusaSlicer 2.4.0 ne demarre pas

Rob91200, 2022-01-07:
bonjour, apres avoir mis de coté l'impression 3D j'y suis reveu et mis a jour PrusaSlicer, mais depuis la mise a jour le logiciel ne veut pas demarré.
la fenetre de demarage se lance puis se ferme et plus rien.

Savez vous comment je pourais regler mon probleme ?
   
Merci d'avance 

Windows 10 
Intel Core 2 Quad CPU Q8200
Ram 6GO

bubnikv, 2022-01-07:
hello, after putting aside the 3D printing I saw it again and updated PrusaSlicer, but since the update the software does not want to start.
the start window starts up then closes and nothing more.

Do you know how I could solve my problem?

thanks in advance

Windows 10
Intel Core 2 Quad CPU Q8200
Ram 6GO

bubnikv, 2022-01-07:
PrusaSlicer 2.4 does not open #7574
prusaslicer 2.4 crashes at start (W32 version) #7613
prusa slice 2.4.0 doesn't start #7560

Rob91200, 2022-01-07:
Merci , je vais regarder ca de plus prêt. Je n'avais rien trouver lors d'une recherche, certainement les mauvais mots clé.

Savez vous ou je pourais trouver une ancienne version du logiciel avec le lanceur, la version PrusaSlicer 2.4.0-beta1 est disponnible ici https://github.com/prusa3d/PrusaSlicer/releases. mais l'installation ne demarre pas, quel technique faut il utiliser pour lancer l'installation ? 

Merci

bubnikv, 2022-01-07:
Thanks, I'll take a closer look. I couldn't find anything during a search, certainly the wrong keywords.

Do you know where I could find an old version of the software with the launcher, the version PrusaSlicer 2.4.0-beta1 is available here https://github.com/prusa3d/PrusaSlicer/releases. but the installation does not start, what technique should be used to start the installation?

Thank you

bubnikv, 2022-01-07:
Maybe you can use https://translate.google.com/ for us?

Kachidoki2807, 2022-01-20:
Salut @Rob91200,
On manque cruellement d'informations pour t'aider. Pour te faire aider, peut-être peux-tu te rapprocher de la communauté francophone du forum ici => [PrusaPrinters](https://forum.prusaprinters.org/fr/).
Reviens nous donner des détails ici s'il s'avère qu'il s'agit d'un bug de PrusaSlicer, ou si on peut simplement fermer ce ticket.
Merci.

Hi Rob91200,
We are lacking information to help you. To get help, maybe you can get closer to the French-speaking community of the forum.
Then, come back here to give us details if this is a PrusaSlicer bug, or if we can close this ticket.
Thank you.

closed 2026-09-29 by prusaslicer-bot[bot]