prusa3d/PrusaSlicer #7709: Ironing in spiral mode

Hurin24, 2022-01-08:
### **Version**
Version 2.4.0+win64

### **Operating system type + version**
Windows 10

### **3D printer brand / version + firmware version (if known)**
Creality Ender 3 + Klipper

### **Behavior**
I had a need to print a waterproof vase in spiral mode. To ensure water resistance, I used a 0.8mm nozzle, 1.6mm wall thickness, 0.4mm layer height. But it is always problematic to make the bottom of the vase waterproof. Or you need to do a lot of layers of the bottom. Or you need to smear the bottom of the vase with dichloroethane. I came up with another way to solve this problem. You just need to do ironing to smooth out all the cracks. But the problem is that it is not possible to smooth only the last layer of the bottom. PrusaSlicer is allowed to do ironing either of all layers, or not at all. If you use ironing on all layers of the bottom, then the bottom is not very beautiful. I understand that this is due to the fact that I use spiral mode, but maybe someone knows how to solve this problem?It is also not possible to use the layer range modifier, as it does not work in spiral mode.

[image]
[image]
Not work
---------------------------------------------------------------------------------------------------------------------------------
[image]
[image]
Not work
---------------------------------------------------------------------------------------------------------------------------------
[image]
[image]
1 layer
[image]
2 layer
[image]
3 layer
---------------------------------------------------------------------------------------------------------------------------------
[image]
The layer modifier does not work in spiral mode

fe60, 2022-01-14:
Instead of ironing the top layer, you can try to print the infill of the second (and the third) layer with a higher extrusion multiplier [1].
You can do this changing the _Flow Percentage_ which is set to 0.95 at the beginning of each .gcode file (search for `M221 S95` in the .gcode file or check the end of the _Start G-code_ in the _Custom G-code_ section of the PrusaSlicer settings) [2].
I propose to increase this value right before the solid infill of the second (and third) layer is printed and to switch back to 0.95 afterwards.
```
...
;TYPE: Solid infill
;WIDTH:0.450175
M221 S105
...
...
...
M221 S95
;LAYER CHANGE
...
```
The horizontal slider and the gcode in the print preview help to identify where the machine codes have to be added to the generated gcode-file.
<details>
<summary>Screenshot showing tool position and gcode at the beginning of solid infill</summary>

[image]
</details>

[1] https://help.prusa3d.com/en/article/watertight-prints_112324/#extrusion-multiplier
[2] https://marlinfw.org/docs/gcode/M221.html

Hurin24, 2022-01-30:
> **I don't know if my proposal is supported by the Klipper firmware your printer is operated with.** Instead of ironing the top layer, you can try to print the infill of the second (and the third) layer with a higher extrusion multiplier [1]. On my Prusa Printer I can do this changing the _Flow Percentage_ [2]. Increase this value right before the solid infill of the second (and third) layer is printed and switch back to the default afterwards.

[image]

fe60, 2022-01-31:
I forgot to explicitely mention, I modified the generated gcode file and added the commands (M221 Sxx) manually using a text editor. You can't apply the proposed solution from PrusaSlicer UI.
I don't know if this will work on your printer, but at least the command M221 seems to be known to your printer.
https://www.klipper3d.org/G-Codes.html#g-code-commands

caufang, 2022-11-16:
I've run into this same problem before and I found that changing "Top fill pattern" and "Bottom fill pattern" to "Concentric" fixes the issue.

I found that generally the Monotonic/Rectilinear fill pattern generally leaves little holes where the solid infill meets the perimeters, and more often than not those holes end up vertically aligning.

By changing the Top and Bottom fill pattern to Concentric, you end up running an extrusion of filament over those holes and that generally seems to stop leakage.

wikolii, 2024-05-11:
I've struck this same issue now with a project I started.
Why does spiral mode only respect "All solid surfaces" mode for ironing is baffling.
I long for the "All top surfaces" mode to work

caufang, 2025-02-07:
@wikolii This is probably because what is considered a top/solid surface is based on if there is more model above that position, not if there are extrusions above.

"All top surfaces" doesn't work because there is more model above those layers, even if there are no extrusions directly above.

closed 2026-09-29 by prusaslicer-bot[bot]