prusa3d/PrusaSlicer #7710: Placeholders for bounding box info

n8bot, 2022-01-08:
I once misled someone in a comment here into believing it was possible to automatically generate and locate a custom g-code prime tower based on bounding box info from placeholders.

I must have been daydreaming about my days using IceSL slicer, which has this ability. I'm now looking again and realize I was completely wrong -- there is no axis extent or bounding box info available to the macro placeholders.

E.g., IceSL has the following placeholders, among others:

| Variable | Type | Description |
|----------|------|-------------|
| `min_corner_x` | float | lower-left corner of geometry to be printer (X coordinate) |
| `min_corner_y` | float | lower-left corner of geometry to be printer (Y coordinate) |
| `extent_x` | float | size of geometry to be printer (X coordinate) |
| `extent_y` | float | size of geometry to be printer (Y coordinate) |
| `extent_z` | float | size of geometry to be printer (Z coordinate) |

I'm sorry if this is a duplicate. I searched, and noticed some mention the total height of the print, etc. but this is more far-reaching. :)

bubnikv, 2022-01-10:
Bed: print_bed_min, print_bed_max, print_bed_size
First
layer: first_layer_print_convex_hull, first_layer_print_min,
first_layer_print_max, first_layer_print_size

n8bot, 2022-01-10:
Lol. What is wrong with my brain!?

Thank you. I really thought I was going crazy! 

I think I got tricked because it seems like non of those are listed here: https://github.com/prusa3d/PrusaSlicer/wiki/List-of-all-placeholders-taken-from-the-code

And also that page could use maybe a few more words to explain things.

lukasmatena, 2022-01-11:
I agree that documentation of the placeholder variables should be improved. The wiki page that you mention is legacy, I just replaced its contents with a link to the one and only documentation that we intend to keep up-to-date. Except it is currently not up-to-date. We'll fix it.

Anyway, I believe you have all the placeholders you need, right?

n8bot, 2022-01-11:
> Anyway, I believe you have all the placeholders you need, right?

Yes it seems that all the placeholders I thought were there are indeed there. I'm not perfectly clear on their use but that's probably due to my bad memory. Thank you.