prusa3d/PrusaSlicer #7711: Height rage modifier triggers "Wipe tower is only supported if all objects have the same variable layer height"

krepsy3, 2022-01-08:
SLIC3R - Prusaslicer 2.4.0

SYSTEM - Windows 10 (I suspect this doesn't play any role)

The problem arises, in my case, in MK3S+ MMU2S print, where I want to add a height range modifier for some objects.

Expected behaviour - I can add any height range modifiers to any of the objects / part groups in the project.

Observed behaviour - if I add the range height modifier to the part group ("Sloučení"), the slicing is refused with an error message "Wipe tower is only supported if all objects have the same variable layer height". I suspect that the application uses part of the same code for adding variable layer height and for dealing with range modifiers. 

What surprises me is that adding the same interval Height range modifier to all the other parts doesn't "synchonize" them internally, in terms of the variable layer height. So this seems like a code loophole, because the height range modifier feature doesn't create variable layer height record for the part it's used on. Though this is all but a speculation coming from my knowledge of programming.

I also tried to "fix" (or overcome) the issue by making a variable layer height on one part and i tried to copy the variable layer height feature from one part to the other, but this doesn't work at all (the variable layer height, even though it's shown as an item in the right hand side TreeView, is tightly bound to it's object and attempt of copying it results in cloning the whole object).

Which also leads me to conclusion, that the error message about wipetower needing the same variable layer height across all objects, isn't really trully fullfilable, unless you group everything together, so that there is only one object to hold the variable layer height modifier, which seems to correctly fullfil the needs of the wipe tower to be generated. 

So, I propose a discussion whether the feature to be able to copy variable layer height object modifier shouldn't be added. That would mean the code behind the error message would need to be changed, and so resolving of this issue should be done in cooperation of adding the mentioned feature, if it's decided that the feature is to be added, to avoid wasting time on fixing this issue in the current situation, if it gets changed immediatelly after.

I'm attaching two .3mf project files directly from Prusaslicer, "psu cover.3mf" is how I discovered the bug. The bug should replicate easily in anyone's Prusaslicer, as I said, this seems like a logical error and code loophole, rather than some weird behaviour only on my PC. 
The second attached file, "psu cover fix.3mf" is for reference only, it shows how I overcame the issue, and shows what I was trying to achieve (it's a MMU print, there are letters and I wanted the first layer of the letters to have full perimeters).

[Prusaslicer issue reference.zip](https://github.com/prusa3d/PrusaSlicer/files/7833975/Prusaslicer.issue.reference.zip)

SimGas, 2022-03-23:
Hi. 
I've experienced exactly the same problem. Even with Prusaslicer 2.4.1 the issue still persists. I've also attached a 3mf file where I discovered the bug.
[Water_buggy.zip](https://github.com/prusa3d/PrusaSlicer/files/8336394/Water_buggy.zip)
.

lukasmatena, 2022-03-24:
Thanks. Yes, this seems to be a bug. Same layer height modifier setting should make the test pass. This should be fixed.

> I suspect that the application uses part of the same code for adding variable layer height and for dealing with range modifiers

Correct. The error message mentions variable layer height, but it fact it checks (or it should) that the layering of the objects is the same.

> Which also leads me to conclusion, that the error message about wipetower needing the same variable layer height across all objects, isn't really trully fullfilable

The purpose of the error is not to be easily fulfillable, but to stop the calculation when a precondition is not met. The only practical way to get the same variable layer height is to copy the object with the layer height already assigned. And of course setting the 'Layer height modifier' the same, as you rightfully assumed.

> So, I propose a discussion whether the feature to be able to copy variable layer height object modifier shouldn't be added

It may make some sense to be able to copy-paste the variable layer height or layer height modifier in some cases, but it also needs to address cases when the objects have different height or have some custom layering assigned already. Both are probably simple to solve, but I personally don't see much added value after the initial problem is fixed. But I may be wrong.

lukasmatena, 2022-03-30:
Also reported in #7903 and #8044 (both closed to track it here only). Other 3MF files to reproduce are available there.

lukasmatena, 2022-04-05:
This should be fixed in the just released 2.4.2-rc1. Can you please confirm? Thanks.

SimGas, 2022-04-07:
> This should be fixed in the just released 2.4.2-rc1. Can you please confirm? Thanks.

I can confirm the issue is resolved in the new release :-)

krugerdave, 2022-04-25:
Hi there, I'm running 2.4.2-rc2 and I seem to be having the same problem still, or at least I think so. I've got two objects on the build plate, one of which has a variable layer height up at the top and the other which I'd like to use for wiping into. Problem is that it still won't let me slice the project while two objects have different layer heights. As for copying the layer heights from one object to another, that sounds like a great idea, but I can't see any way to do it. Am I missing something? Thanks!