prusa3d/PrusaSlicer #7713: Gcode from version 2.4.0 but not 2.3.3 causes printer error

Heindal7, 2022-01-09:
### Prusa Version
2.4.0 but not 2.3.3

### Operating system type + version
Windows 10 Pro 21H1

### 3d Print Server
Repetier Server 1.2.1 on a Pi Zero W

### 3D printer brand / version + firmware version (if known)
Anycubic Vyper 2.3.3 and 2.4.5

### Behavior
Printer stops mid print by traveling straight up in the Z direction about 20mm and stops responding to further G-code.
The issue doesn't not occur in all prints. When I repeated a print with the issue, it also happened on the second run at roughly the same location. It also repeated when I re-uploaded it to the server.
It has always happened (all six times) near the a perimeter of a hole and doesn't occur after the same duration of time on different prints. It happened on layer 14 of the printing of the attached file at about line 11868.
[tibia_l_0.2mm_TPU_33m.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/7834459/tibia_l_0.2mm_TPU_33m.gcode.zip)
I have stopped using version 2.4.0 and have not had any issues on other slicers.

neophyl, 2022-01-09:
Only thing I see in that location is the following for the first colour change.
; AFTER_LAYER_CHANGE 13 @ 2.9mm
;COLOR_CHANGE,T0,#21DA30
M600

Try editing out the # from there and printing.  Its possible your printers gcode parser cant handle that character.

rtyr, 2022-01-09:
> Printer stops mid print by traveling straight up in the Z direction about 20mm and stops responding to further G-code.

Isn't that just normal behavior of your printer after M600 command? 

Maybe you just added color change accidentally?

Heindal7, 2022-01-09:
I followed your suggestion and moved the M600 command earlier in the file and that triggered the issue. That solves part of the issue even if I now know the color change command is buggy on my printer.

I rechecked the stl files that had the problem. Sometime when it finds a file that it thinks is a logo, it appears that the slicer is automatically adding color change to my prints when if I don't select the apply color change automatically. Worse than that once it happens, it appears to think all future files I load are also logos (even if it doesn't normally) and continues to apply the color change to them. If I close and reopen the program, it stops thinking everything is a logo and goes back to doing everything as a single color. 
[image]

bubnikv, 2022-01-10:
PrusaSlicer will only apply the automatic "color change" to your model if you click on the hyperlink "Apply color change automatically".

lukasmatena, 2024-02-19:
Closing. There are no similar reports, it could really have been caused by inadvertent addition of a color change.

closed 2024-02-19 by lukasmatena