prusa3d/PrusaSlicer #7714: Empty G1 before starting bridging perimeter

Grawp, 2022-01-09:
### Version
Version 2.4.0

### Operating system type + version
NixOS, nixpkgs#ff377a78794d412a35245e05428c8f95fef3951f

### 3D printer brand / version + firmware version (if known)
MK2.5S 3.10.0-142

### Behavior
'Empty' (i.e. without parameters) G1 command is emitted before starting some of bridging perimeter.

[image]

#### Project File (.3MF) where problem occurs
[bridge.zip](https://github.com/prusa3d/PrusaSlicer/files/7834841/bridge.zip)

Grawp, 2022-01-09:
This may be related to or even causing https://github.com/prusa3d/PrusaSlicer/issues/1284#issuecomment-1008272202 !
Perhaps the G1 commands should be there and contain some crucial step preventing beginnings of bridging perimeters becoming too thin or outright starting in mid-air whole millimeters after edge.

bubnikv, 2022-01-10:
From https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-beta2 change log:

> Empty G1 commands were sometimes present in the G-Code due to the cooling slow down logic. This is an old bug present in 2.1.0 already #3058 #5176

bubnikv, 2022-01-10:
The previous occurence of empty G1 was fixed with https://github.com/prusa3d/PrusaSlicer/commit/4e9f906848f6bbe31476e355919692ec1fb7b40d

bubnikv, 2022-01-10:
This one was fixed with 07d0484e32fb71145934025fd25c2e08dd9bd395

closed 2022-01-10 by bubnikv