prusa3d/PrusaSlicer #7716: Sending gcode to repetierserver on network cancelt with "Error 65" PrusaSlicer V2.4

danxde, 2022-01-09:
PrusaSlicer V2.4  on macOS 12.1.  

I use a repetier-server 1.2.1 with an MK3S+ (latest firmware)

If I try sending the G-Code direct to the repetierserver,  the following  error occured:

"Fehler beim Hochloden zu Druckhost:
Send failed since rewinding of the data stream failed:
Cannot rewind mime/post data
[Error 65]"

This issue is only in the new version.  With PrusaSlicer 2.3.3 this error not occured, testet with  Down-/Upgrade between both  versions.

[Dyson Knopf.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7835247/Dyson.Knopf.3mf.zip)

bubnikv, 2022-01-10:
@repetier Would you guys help, please?

danxde, 2022-01-10:
I have now tested it with another project, the upload to the server also works with version 2.4.  Maybe it's only a single case.

repetier, 2022-01-10:
I also just tested and it worked. Got similar error a few days ago, but that was just ip not reachable and hence not able to upload. So maybe wifi on your server pi was just having troubles when you tried and now it is back connected. Next try check pi is reachable and retry on slicer then.

bubnikv, 2022-01-10:
@repetier
Thanks. I think we can close this issue.

bubnikv, 2022-01-10:
CURLE_SEND_FAIL_REWIND (65)

When doing a send operation curl had to rewind the data to retransmit, but the rewinding operation failed.

bubnikv, 2022-01-10:
https://curl.se/libcurl/c/libcurl-errors.html

closed 2022-01-10 by bubnikv

danxde, 2022-01-10:
Hi, the Server was reachable (Rasp-Pi4 with LAN connectivity), at the same time I was also   connected to the Web-GUI.  Also I tried multiple times to send the gcode with the slicer.  If the issue is  repeating, I will take a deeper look in the connectivity.      

Thank's  for responding/checking!

ksvladimir, 2022-01-22:
I found a way to fix it: it works when I explicitly include `https://` in the hostname:

[image]

If I omit `https://`, the "Test" button says "Success", but uploading files never works (even though network is fine and I can upload files using curl). I suggest to re-open this bug and fix it by either failing the "Test" when `https://` is not there, it making upload work without it.

dragnmastr85, 2022-01-27:
> I found a way to fix it in my case: it works when I explicitly include `https://` in the hostname:
> 
> [image]
> 
> If I omit `https://`, the "Test" button says "Success", but uploading files never works (even though network is fine and I can upload files using curl). I suggest to re-open this bug and fix it by either failing the "Test" when `https://` is not there, it making upload work without it.
> 
> I'm using PrusaSlicer 2.4.0 on Apple M1 and OctoPrint 1.7.3 on R-Pi4.

Thank you!!

pridkett, 2022-03-07:
I recently started getting this problem out of no where too. No changes to my network. Raspberry Pi running OctoPi has not changed. Certificate is a real certificate from ZeroSSL issued via Acme.sh. Issued on January 29th, 2022 and valid until April 29th, 2022. Printing worked fine until around March 3, 2022. No CA file is needed because it's an SSL certificate with a chain of trust back to a certificate authority.

Adding in `https://` to the beginning of the hostname, as [described by @ksvladimir seems to fix the problem](https://github.com/prusa3d/PrusaSlicer/issues/7716#issuecomment-1019362880).

Doing some more digging shows that there is something weird that happened with how OctoPi/OctoPrint are now presenting the SSL certificates - as a straight curl command does not work anymore either. Other certificates fetched by Acme.sh work fine.

```
$ curl https://octopi.MY.FQDN/
curl: (60) SSL certificate problem: unable to get local issuer certificate
More details here: https://curl.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.
```

What's more interesting is that the certificate shows up as being appropriately trusted in browsers across multiple devices and operating systems. Seems like it's an OctoPrint or OctoPi bug.

rvalle, 2023-01-22:
> I found a way to fix it in my case: it works when I explicitly include `https://` in the hostname:

Thanks @ksvladimir !! this has been torturing me for ages.

I don't understand why won't they fix this. it is really annoying.