prusa3d/PrusaSlicer #7719: X and Y compensation

Digibike, 2022-01-09:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Digibike, 2022-01-09:
Hi,

in PrusaSlicer is a function to compensation X/Y. There are 2 problems with it. The first and little is, it works with this Numerical value in X and Y. If this are different, this works only as an Compromise.
The second one is the greater problem: If I had different holes - circular, oval or angular, it will be smaller in 3D Print than in CAD. I can consider it in CAD, but than I must know, with which Nozzle I will later Print. If I Change later the Nozzle, I must correct it in CAD and Export it again. Not so good workflow, because the mistake is not in CAD, but I must correct it in CAD...
If there is a function to correct holes, or I can give not an absolute correction value but a % value, so I can enlarge it and at the same time reduce it again with scaling, so that ultimately only the holes are enlarged accordingly, but not the outer contours themselves ... As an example, compensation -5% reduces the outer contours and then scaling with 105% increases the area in itself. 
I hope, i had explained, what I mean...

With regards, Digibike

neophyl, 2022-01-10:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/515 I think.

stepikovo, 2022-05-26:
I agree, seems like a duplicate of #515, closing.

closed 2022-05-26 by stepikovo