prusa3d/PrusaSlicer #7720: (Regression TBC) Seam alignment REAR vs Seam painting

Moyrn, 2022-01-09:
### Version
2.4.0

### Operating system type + version
Win10 64

### 3D printer brand / version + firmware version (if known)
MINI

### Behavior
Like in 2.4.0 alpha 1 (but not in an exact same way), rear seam conflicts with seam paint
In 2.4.0 alpha 3 it is OK - same m3F file
So unsure this is actually a regression.
Just sharing...

#### Project File (.3MF) where problem occurs
[MANCHON-SUNLU-MINI-B25.zip](https://github.com/prusa3d/PrusaSlicer/files/7835978/MANCHON-SUNLU-MINI-B25.zip)

Seam paint
[image]

2.4.0 
[image]

2.4.0 alpha 3
[image]

AnatiliyKsl, 2022-01-13:
1 lately, all programs with such an effect (this is sad)-Slic3r , SuperSlic and so on,. 
The seam position is at the back. In the beta version, the seam is obtained strictly vertically. In version 2.4.0, the seam on some layers is offset from the vertical, as in the previous post. If you set it - print the outer perimeter first - everything is fine, but the print quality becomes worse, the outer perimeters are obtained with defects.
2 In version 2.4.0, the zoom control works in reverse, ??????

AnatiliyKsl, 2022-01-13:
[image]

AnatiliyKsl, 2022-01-14:
Version
2.4.0

Win 10  x64

If you turn off: Print settings - layers and perimeters - quality - thin wall detection, the seam is obtained vertically. Print outer perimeters first - off.

Moyrn, 2022-01-14:
Hello. Thanks.

Meanwhile i moved to nozzle 25, and the scar is less scary ;-)
So with or without "avoid thin walls" it stays vertical. not straight, but
vertical
And I avoid the "thin walls" option as even in nozzle 25, some of my
printed letters on the surface are not printed at all. Disabling the option
makes them appear again.

Here's on another object, in 2.4.0
[image: image.png]

closed 2026-09-29 by prusaslicer-bot[bot]