prusa3d/PrusaSlicer #7721: [Feature Request] Open DWC after sending a code button

kr15uk, 2022-01-09:
[Feature Request]

Duet Web Control URL button after "upload and print"

bubnikv, 2022-01-10:
Would you please provide more detail?

kr15uk, 2022-01-10:
Sorry, intention was to keep it as short as possible to save your time reading memoirs.
Also looks like I've negated to mention DWC is short for Duet Web Control, sorry for that.

Sure thing.
1. Slice the file
2. Click on "Send G-Code to printer host"
3. Upload and Print
4. After upload is done you have "COMPLETED" and X to dismiss the message (see below)
[image]
 
Would be super handy to have another button with "open URL" for the machine.
Sorry for comparisons but Cura does have this and when I have to use Cura I find that extremely helpful.

Ta

Kris

bubnikv, 2022-01-10:
> Sorry for comparisons but Cura does have this and when I have to use Cura I find that extremely helpful.

I suppose Cura itself does not have it, but some plugin. Which one are you using?

bubnikv, 2022-01-10:
> Oh you're right on this, it is a plugin by the looks of it, sorry for that.
> It's been automatically synced to any new install via Ultimaker account
> sign in so I've assumed it's part of Cura itself.
>
> Never the less, very handy to have. I've included a picture of how it
> looks when sliced and sent over to the machine.
> [image: Screenshot 2022-01-10 at 13 27 20]
> <https://user-images.githubusercontent.com/5671041/148773701-9d025384-9c9b-4c28-bf64-c67cd943f2e5.png>
>
> Just for the future record and/or for whom this might be useful, it's
> called "Duet RepRap Firmware Integration" on Cura's Marketplace thing where
> you get your plugins/material profiles etc.
> Also here - https://github.com/Kriechi/Cura-DuetRRFPlugin

We were discussing a similar feature for our printers, however we were
unable to solve logging in into the web service. Does the Cura plugin take
care of it, or do you have to log-in manually?

Do you have to enter the web page address  in addition to the API IP
address? Or is the web page address derived from the IP address?

kr15uk, 2022-01-10:
It's on isolated workshop network so never bothered to set up access control, I know not very wise but oh well.

In my case all I need is URL of the machine but by the looks of it it also handles authentication (please see screenshot of config window)

[image]

Hope this helps.

Kris

bubnikv, 2022-01-10:
Thanks. I see there is no specific http address entered for the WEB
interface. I suppose Open Browser just opens the browser with the Duet
Address (URL) field filled in.

closed 2026-09-29 by prusaslicer-bot[bot]