prusa3d/PrusaSlicer #7722: PrusaSlicer 2.4 crashes on start up - MacOS Catalina x86

dtackgrounded, 2022-01-10:
### Version
2.4.0+x64

### Operating system type + version
MacOS 10.15.7 Catalina

### 3D printer brand / version + firmware version (if known)
Stock Prusa Mini
CR-10 Mini
Ender 3 v2

### Behavior
PrusaSlicer crashes on start up.
I have traced it to the default start up printer being the Creality CR-10 Mini.
Running PrusaSlicer 2.3 has no issues. If I exit PrusaSlicer 2.3 with the last printer used as CR-10 Mini, starting PrusaSlicer 2.4 will cause it to crash. If I exit PrusaSlicer 2.3 with Prusa Mini as the last printer used, PrusaSlicer 2.4 will start correctly. I can switch to my Ender 3 v2 config in 2.4 with no issues, but selecting CR-10 Mini as my target printer in 2.4 will cause the App to crash.

I have attached a screenshot of the crash page; it appears to have crashed while trying to render the CR-10 Mini printer bed.

[image]

FidelCapo, 2022-01-10:
Can you please attach your zipped configuration folder? (Help -> Show Configuration Folder). Thank you

dtackgrounded, 2022-01-11:
I reset my Printer profiles and started fresh using the 2.4 Configuration Assistant; it nuked all my previous settings, but now no longer crashes selecting the CR-10 Mini. It must have been some backwards compatibility issue with 2.3.

Nuking my 2.3 settings was required because having run 2.4 once, 2.4 altered some of the config files which were not compatible with 2.3 - gcode_flavor = marlin2. After 2.4 updated my config, 2.3 would no longer start so I was stuck in the middle and took the drastic measure to just reset everything.

closed 2022-01-11 by dtackgrounded