prusa3d/PrusaSlicer #7723: Multicolor print slicing changes to worse

wlcina, 2022-01-10:
### Version
2.4.0

### Operating system type + version
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
Does not matter, but 3x Prusa MK3S+ with MMU2S FW 3.10

### Behavior
MMU2S slicing is different than before which causes lot of issues in my model database.
When the model parts overlap, currently (in 2.4.0) there is no control which model part will be printed (while skipping the overlapped part of another model part). In 2.3.3 - slicer gave priority to the model part which were higher in the list of the model parts. 
An option to return to old behavior would be great, I am attaching screens, models are the same, just drag and dropped to both of slicers:

Prusa 2.3.3:
[image]

Prusa 2.4.0:
[image]

bubnikv, 2022-01-10:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7482
See discussion in https://github.com/prusa3d/PrusaSlicer/issues/7482

lukasmatena, 2024-02-19:
Closing as a duplicate of #7482.

closed 2024-02-19 by lukasmatena