prusa3d/PrusaSlicer #7724: Prusaslicer 2.4.0 crashes to desktop

renaatski, 2022-01-10:
### Version
Version: 2.4.0
Build: PrusaSlicer-2.4.0+UNKNOWN

### Operating system type + version
Version: 2.4.0
Build: PrusaSlicer-2.4.0+UNKNOWN
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 34,268MB

GPU = AMD RX6800XT
Was running Adrenalin 21.10.1,  and even updated to 21.12.1 to see if it made a difference.

### 3D printer brand / version + firmware version (if known)
MK3S+

### Behavior
PrusaSlicer startup either hangs or crashes to desktop. Sometimes it does start properly (especially after a reboot), but as soon as I click to open a file (project or import STL) , it crashes to desktop again.  I'm not even clicking a file yet, just the file open dialog is enough to crash the app.

Reproducing is simple.
- install prusaslicer
- start prusaslicer
- crashes can be observed immediately

Note: After a fresh reboot, I observe similar issues with file dialog windows crashing PrusaSlicer 2.3.3 to the desktop as well :-/     subsequent tries to launch the app just crash to desktop upon starting prusaslicer immediately.

This is a well maintained issue-free PC thus far, so I'm quite puzzled about where all this is coming from.   I do need the slicer to bee able to use my printer!

I've observed less to no issues when opening prusaslicer over an RDP session from e.g. my iPad.  Not really all that workable of course that way.
 
 * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
         :-)  prusaslicer just starting and ready to go?
* _Actual Results_
  Nothing to show, crash to desktop or hang of the app.
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
no

bubnikv, 2022-01-10:
Should be fixed on master.

renaatski, 2022-01-10:
> Should be fixed on master.

Interesting, is there a nightly build or something that I can test?

bubnikv, 2022-01-10:
Build: PrusaSlicer-2.4.0+UNKNOWN indicates that you self compiled.

bubnikv, 2022-01-10:
prusaslicer 2.4 crashes at start (W32 version) #7613

PrusaSlicer 2.4 does not open #7574
prusa slice 2.4.0 doesn't start #7560

bubnikv, 2022-01-10:
https://github.com/prusa3d/PrusaSlicer/issues/7574#issuecomment-1004701590

renaatski, 2022-01-10:
> Build: PrusaSlicer-2.4.0+UNKNOWN indicates that you self compiled.

not really, I'm installing Visual studio community 2022 as I type this.
I may have tried a .DLL from one of the other issues, but found it wasn't fixing things either for me just yet.

lukasmatena, 2022-01-10:
@renaatski In that case I would be curious where the UNKNOWN build was from, who built it and with what modifications. Anyway, you can try this build: https://slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0+29-win64-g8636ec8b4-202201100014.zip. You may need to enter
```
user: prusaslicer
password: slicer
```

 when asked. Please, let us know if you still have issues with this build.

renaatski, 2022-01-10:
> @renaatski In that case I would be curious where the UNKNOWN build was from, who built it and with what modifications. Anyway, you can try this build: [...]

Thanks for that. I downloaded it and ran it.  So it doesn't hang or crash on startup now anymore...

However... I still have issues with opening files, the windows file dialog appears briefly, then freezes and crashes the app to desktop after  2 seconds :-o

renaatski, 2022-01-10:
I may have found something... I had a NAS shared drive mapping that was unavailable (NAS turned off), so with a red cross. I just removed it and now the dialog does no longer crash the prusaslicer now?   Will keep on testing

lukasmatena, 2022-01-10:
@renaatski That sounds like a separate issue to me. Can you please test whether the same happens with [PrusaSlicer 2.3.3](https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.3.3/PrusaSlicer-2.3.3+win64-202107161027.zip)? Thanks.

bubnikv, 2022-01-10:
You also may have the dreaded Shitubox issue.
https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file

renaatski, 2022-01-10:
I spoke too soon.. Still having crash to desktop when trying to import an STL file. The dialog that's supposed to come up crashes the app. It was unrelated to the NAS drive (that is now gone ofc)

renaatski, 2022-01-10:
I don't have shitubox.  I even searched the filesystem for *chitu*  ...

renaatski, 2022-01-10:
[image]

so weird, now again, it hangs upon startup? pfff...

renaatski, 2022-01-10:
kinda weird because this pc is self-built, well maintained and not that old of an install either.  I don't have any issues with other programs that I'm using which makes this even stranger

lukasmatena, 2023-02-08:
Do you still see this issue with [2.6.0-alpha3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha3)?

renaatski, 2023-02-08:
I stopped having this issue when I re-imaged my PC from windows 10 to
windows 11. So a clean install.   Never saw this problem again

Kind regards,

Renaat

lukasmatena, 2023-02-08:
Thanks for getting back to us. We also did some changes in the meantime which might help with this, so I think it is safe to close this as resolved.

closed 2023-02-08 by lukasmatena