prusa3d/PrusaSlicer #7727: Something wonky with Variable Layer Height

bbinnard, 2022-01-10:
### Version
2.4.0

### Operating system type + version
Windows 11/64 Home

### 3D printer brand / version + firmware version (if known)
FLSun SuperRacer with Marlin 2.0.8.1 and TFT70 LCD screen

### Behavior
Variable Layer height produces extremely tiny layers at the start of a print

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[image]
[image]

I wanted the first few layers to be shorter than my default layer height of 0.200 mm to minimize overhang and ensure a smooth surface. So I made the variable layer height settings look like this:
[image]

What happened was the actual layer height was extremely small and it took many hours to print the first few mm:
[image]

This shows that after nearly 14 hours of printing (13:58:29) the print has reached a height of only 9.79 mm. I haven't looked at the GCode generated for the print because the SD card is in the printer now, but the first set of layers were VERY thin, my guess is around 0.005 mm or so. It seems to me there should be some sort of lower limit to the minimal layer height generated by the variable layer height function.

n8bot, 2022-01-10:
You need to change the minimum layer height setting in your printer profile. It defaults to a minimum of 10 microns if no other value/0 is set. 

Also, the gcode you are looking at is saying the layer thickness. It chose a 0.0700158 mm layer height. You might have clicked the "smooth" button which will smooth the layer height translations and lead to those funny looking numbers. The G1 Z move is the absolute Z height that the nozzle is printing at. 

TL;DR change your minimum layer height setting.

bbinnard, 2022-01-10:
Thanks for the prompt reply. My min. layer height is set to 0.07 which is the default for Ver. 2.4 So I'll change it to 0.10 which hopefully will be OK. 

Yes, I did click the Smooth button. I guess I won't do that again.

lukasmatena, 2022-01-11:
If this is solved, please, close the issue.

If this is not solved, please, provide a 3MF.

Thanks.

bbinnard, 2022-01-11:
Apparently the problem had to do with the smoothing function which I had not used before.  So I'll just avoid it in the future.

closed 2022-01-11 by bbinnard