prusa3d/PrusaSlicer #7728: Bridge infill settings

scotfor, 2022-01-10:
Version: 2.4.0-alpha3+win64
Build: PrusaSlicer-2.4.0-alpha3+win64-202109301506
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 

Printers are Kingroon KP3 and Geeetech A10M.

A known problem discussed in on-line forums.

There are no settings for "bridge infill".
There needs to be since it often does not print well with the preset, hidden settings.
** If nothing else it would be good to be able to just turn it off so "solid infill" could be used.

The problem I see most often is a lack of bridging - gaps in the layer over a grid infill ( 20% for example )
A couple of possible problems - 1) Speed too high, 2) Low extrusion.

The slicer image won't help since all looks fine there.

lukasmatena, 2022-01-11:
To be honest, I am not sure I understand what you are talking about. A 3MF, a screenshot, or at least a link to where this is "discussed in on-line forums" might help.

As far as I know, solid layer above 20 % Grid infill is printed as bridging. This may also be a duplicate of #569, but you really should expand on it a bit.

scotfor, 2022-01-11:
> To be honest, I am not sure I understand what you are talking about. A 3MF, a screenshot, or at least a link to where this is "discussed in on-line forums" might help.
>
> As far as I know, solid layer [...]

ok.
Links:
https://forum.prusaprinters.org/forum/prusaslicer/bridging-infill-settings/
https://forum.prusaprinters.org/forum/prusaslicer/turn-off-bridge-infill/
https://www.reddit.com/r/prusa3d/comments/ko1ex5/bridge_infill_picks_the_worst_orientation/
https://www.reddit.com/r/3Dprinting/comments/jg37xl/comparing_cura_to_prusaslicer_bridge_infill_issue/
https://github.com/prusa3d/PrusaSlicer/issues/3225
https://3dprinting.stackexchange.com/questions/18163/bridge-infill-does-not-stick-to-the-perimeter

Slicer:
Here is a screenshot of the layer window.
[image]

The most important point of this issue is that the "Feature type"  'Bridge infill' appears and there are no setting for that feature type. None. This is just wrong from a usability, 3d print configuration, and tool design perspective.

The default hard coded values don't work. Here is a pic of the Bridge infill layer I just printed.
[image]

lukasmatena, 2022-01-11:
@scotfor Thanks. Can you please send the 3MF of that print?

scotfor, 2022-01-11:
> @scotfor Thanks. Can you please send the 3MF of that print?

Gladly, but the platform doesn't let me attach 3mf files to this post. How do I post/send it to you?

lukasmatena, 2022-01-11:
@scotfor Rename it to zip, for example.

scotfor, 2022-01-11:
@lukasmatena 
[test peice v4.zip](https://github.com/prusa3d/PrusaSlicer/files/7848693/test.peice.v4.zip)

FloBirni, 2022-01-23:
Instead of opening a new ticket, it might be more helpful contributing to this one. I'm currently facing the issue, that the bride support is visbile from a cutout on the bottom of a printed part, although these cutouts are rather tiny and there's definitely no support layer necessary in my opinion. Therefore, I would really like to get rid of this bridge support layer, because it looks like a settings or print failure. I was searching for quite some time now and it appears, as also stated by the links the author of this issue provided, that there is no possibilty right now to turn the bridge layer off although it would be desired by a lot people.

closed 2026-09-29 by prusaslicer-bot[bot]