prusa3d/PrusaSlicer #7732: Support Cubic request

jzagaja, 2022-01-11:
If top=0 please make option to not put dense infill below top that not exist.

n8bot, 2022-01-11:
Can you show a photo of what you mean?

lukasmatena, 2022-01-11:
@jzagaja Please, post a 3MF. We want to develop software, not play guessing games. Thanks.

jzagaja, 2022-01-11:
Enclosed.

[test_support_cubic.zip](https://github.com/prusa3d/PrusaSlicer/files/7848509/test_support_cubic.zip)

lukasmatena, 2022-01-11:
I don't see an issue. You turned off top layers, you got no top layers. You did NOT turn off infill, so it is still there. Turn it off if you don't want it. Slicer cannot guess what you do or do not want, and the solid infill is important for rigidity of the model, not just for supporting the top layers. Or maybe I misunderstood the problem.

lukasmatena, 2022-01-11:
There are no top supports, there is internal infill. Works as designed.

jzagaja, 2022-01-11:
Isn't support cubic infill to support top layers? This is good idea but you can't pour resin switching top off.

n8bot, 2022-01-11:
What are you trying to accomplish? You want to pour resin inside an object with support cubic infill?

Can you drill a hole after?

jzagaja, 2022-01-11:
Print large objects fast with low amount of filament then pour resin or concrete. Natural choose is gyroid but with low value it wont support top layers. Drilling hole is acceptable for resin but difficult for concrete. Your proposal?

viper0078, 2022-01-11:
Set Infill Density to 0%.

jzagaja, 2022-01-11:
Not good idea (tested) - always fails.

n8bot, 2022-01-11:
> Natural choose is gyroid but with low value it wont **support top layers**. Drilling hole is acceptable for resin but difficult for concrete. Your proposal?

You said no top layers.

jzagaja, 2022-01-11:
No top **highest** layer to pour resin/concrete. Modifier to keep other top layers.

n8bot, 2022-01-11:
Ok, so then why do you want the dense supports removed? For faster printing?

jzagaja, 2022-01-11:
Real example. 

[image]

viper0078, 2022-01-11:
Probably the closest you can get, though, is to use the 'Only Infill Where Needed' option.
Even then, the part marked 'modifier to print top ' will not print.
I think you should change the design of your 3D model instead of asking the slicer to do everything for you.
Are you using Fusion360, and could you use the Shell feature?

n8bot, 2022-01-11:
PrusaSlicer now deals with the overlapping shells method of modelling, like used in some RC airplanes. That might be a good method to employ here to generate whatever support pattern one wishes, with the walls they wish as well.

jzagaja, 2022-01-12:
Jan Herich developed slicer for that: https://instagram.com/jan_herich?utm_medium=copy_link

I use Solidworks and of course know she'll function ;-)

lukasmatena, 2022-01-12:
To be honest, I probably still don't understand what is the request about and what @jzagaja wants us to do. But it sounds as some task not really belonging to slicer. I'm inclined to close the issue unless someone comes with something enlightening.

viper0078, 2022-01-13:
Is this not enough?

[image]
[image]
[image]

viper0078, 2022-01-13:
Is this not enough?

[image]

[image]

[image]

[image]

closed 2026-09-29 by prusaslicer-bot[bot]