prusa3d/PrusaSlicer #7733: Concentric supports request

jzagaja, 2022-01-11:
Please add concentric supports like in Cura. It's easy to remove.

lukasmatena, 2022-01-11:
Concentric support interface pattern is available in 2.4.0. Does it satisfy the request?
[image]

jzagaja, 2022-01-11:
Like this? https://youtu.be/g3jm3YGa3mo

lukasmatena, 2022-01-11:
Like implemented in PrusaSlicer 2.4.0, that anyone including you can download from https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0 and try out. I don't know what I'm supposed to see in that video.

jzagaja, 2022-01-11:
Isn't concentric in Cura all way up not just before touching supported surfaces as in PS 2.4.0?

n8bot, 2022-01-11:
You could abuse the support interface by forcing it to generate like 1000 layers. Then make it spaced far enough that it is no longer dense.

lukasmatena, 2022-01-12:
> Isn't concentric in Cura all way up not just before touching supported surfaces as in PS 2.4.0?

Does it make a difference if ease of removal is the goal?

bubnikv, 2022-01-13:
If you want more stable supports, enable the sheath.

bryn51, 2024-07-20:
concentric supports have already been deployed in PS. This issue should be closed.

closed 2026-09-29 by prusaslicer-bot[bot]