prusa3d/PrusaSlicer #7734: Saving Dialog: Values are truncated

aclucca, 2022-01-11:
### Version
Version: 2.4.0+win64
Build: PrusaSlicer-2.4.0+win64-202112211614

### Operating system type + version

Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19044), 64-bit edition
Total RAM size [MB]: 10,600MB

### 3D printer brand / version + firmware version (if known)
Prusa MK3

### Behavior

Saving Dialog: Values are truncated - see screenshot

[image]

FidelCapo, 2022-01-12:
This is not a issue. The text is truncated because custom gcodes are usually very long. Right-clicking on the text will open a dialog with the full text.

closed 2022-01-12 by FidelCapo

YuSanka, 2022-01-12:
@aclucca , thanks for your observation :) 
An information line was lost after some big code refactoring. Regularly this dialog looks like this:
[image]

This bug is fixed and will be a part of upcoming patch release.