prusa3d/PrusaSlicer #7736: Unexpected Object on Start-Up of PrusaSlicer 2.4.0 in MK3S MMU2 mode

SlipPlane, 2022-01-12:
### Version
PrusaSlicer 2.4.0

### Operating system type + version
Windows10 Pro, OS build 19042.1415

### 3D printer brand / version + firmware version (if known)
MK3S MMU2
Firmware: 3.10.1-4697 & 1.06-372

### Behavior
When I run PrusaSlicer 2.4.0 using "MK3S MMU2" mode I get a weird structure on the screen (top right in image below) at start-up, or, when I switch over to that mode. If I toggle to "MK3S MMU2 Single", then it disappears.  The object does NOT get included in gcode generation but I cannot get rid of it. I have tried this on two different PCs (same O/S) and I get the same result.  
**NOTE:** I did not remove the older version of PrusaSlicer (2.3.3) when installing, nor have I done so.

I have scoured the web & Prusa Forum with no luck in others experiencing this issue....I fear it is just me!!

Image
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
This occurs on start-up, no project/import has been performed.

neophyl, 2022-01-12:
its the purge block

lukasmatena, 2022-01-12:
https://github.com/prusa3d/PrusaSlicer/issues/4848
Intentional, turn it off in Print Settings->Multiple extruders if you don't want it. 
Closing.

closed 2022-01-12 by lukasmatena

SlipPlane, 2022-01-12:
Brilliant!

Many thanks for the incredibly quick reply/solution, I would have NEVER
found it otherwise. It is the first time I have run into the 'problem'.

Mark