prusa3d/PrusaSlicer #7739: Feature Request: First layer only toggle

alexwhittemore, 2022-01-12:
### Version
2.4.0

### Operating system type + version
macOS 11.6, but applies to all OSs

### 3D printer brand / version + firmware version (if known)
Prusa Mk2.5s, but applies to all printers

### Behavior
***Feature request!***

I find, very often, that I'd love to do a first layer test of *this particular* print, to verify that things are going to go smoothly when trying a new filament, new settings, or just challenging geometry. For example: I'm trying to print in nylon, and the extreme temperatures mean that my live Z starts off correct, but then drifts by the time the first layer is done. Or, live Z is fine, but my nozzle temp with a steel nozzle is insufficient. Etc etc etc, we all know the zillion ways a first layer can go wrong.

My point here is, for a challenging or high-stakes print, I very often want to inspect the first layer for fitness, or even pull the first layer off the printer for inspection before I actually print the model. The built-in M601 functionality works pretty well for this, to insert a pause before the second layer. But I'd LOVE to just have a toggle somewhere in print settings that I could leave on as a default in certain print modes to ALWAYS pause after the first layer.

alexwhittemore, 2022-01-12:
Can you? Is there any logic for "only insert this code after the specified layer count"?

Even if so, I'd argue that the presence of such a toggle would be a "hey that's a useful process idea!" But then if literally all the "requirements" can be satisfied manually, I'd understand not wanting to add clutter.

Grawp, 2022-01-12:
> Can you? Is there any logic for "only insert this code after the specified layer count"?
> 
> Even if so, I'd argue that the presence of such a toggle would be a "hey that's a useful process idea!" for users. But then if literally all the "requirements" can be satisfied manually, I'd understand not wanting to add clutter.

https://help.prusa3d.com/en/article/insert-pause-or-custom-g-code-at-layer_120490/

Grawp, 2022-01-12:
You could insert conditionally generated M601 to _Before layer change_ and save it as a new printer profile.

alexwhittemore, 2022-01-12:
Ahh, now that I've gone looking for it, I see that there's a whole language available for conditional gcode processing, and it looks like `layer_num` is the variable you'd use.

Ok, at this point I admit that everything I'd like CAN be accomplished manually. Of course, I still like the relative simplicity of a toggle in basic print settings, and I think others (especially newcomers) would benefit from it. But I allow that it might not pay its way in the UI.

lukasmatena, 2022-01-12:
> how to implement this workaround idea.

It should be only about adding `{if layer_num==1}M601{endif}` into the 'before layer change' custom gcode.

> the presence of such a toggle would be a "hey that's a useful process idea!" for users

It surely would, but the problem is (as you have guessed) that different people have dozens of similar useful ideas, and the number of settings and switches is already too high. We are generally against adding configuration options, unless it is really justified. Also, everything that we add must stay, removing the options would break existing profiles. We are trying to make a software that does not need to expose every imaginable parameter for the user to tweak.  There are forks of PrusaSlicer whose authors have different opinion on this, which is completely fine.

closed 2026-09-29 by prusaslicer-bot[bot]