prusa3d/PrusaSlicer #7744: Fuzzy Skin Overhangs

fabiobollinger, 2022-01-14:
### Version
Version 2.4.0+win64

### Operating system type + version
Windows 10 Pro 20H2

### 3D printer brand / version + firmware version (if known)
Stock Prusa i3 mk3s - firmware not related to this issue

### Behavior
Reproduce
>In prusaslicer 
>Imported STLs of custom chess pieces
>Custom print settings confirmed to make good parts
>Experimenting with "Fuzzy Skin"
>Fuzzy Skin: "Outside Walls" 
> thickness: 0.08mm
> distance 0.8mm
> slice
> sharp corners in models overextend way too much, see screenshots:

[image]
[image]
[image]

* _Expected Results_
See the model. The corners are sharp but they do not protrude
[image]

* _Actual Results_
See previous screenshots - fuzzy corners protrude way more than expected
_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
Github is not allowing me to upload .3MF - if you like I can upload as TXT and you change the file type on your end. let me know

lukasmatena, 2022-01-14:
You can zip the 3mf and upload it. Thanks for the report.

fabiobollinger, 2022-01-14:
Hi Lukas, thanks for the tip! here you go.
[All Pieces.zip](https://github.com/prusa3d/PrusaSlicer/files/7868326/All.Pieces.zip)

closed 2026-09-29 by prusaslicer-bot[bot]