prusa3d/PrusaSlicer #7745: Movement not optimized

RudolphRrr, 2022-01-14:
### Version
2.4.0

### Operating system type + version
Windows 10 64 Bit

### 3D printer brand / version + firmware version (if known)
not relevant

### Behavior
The path is anything but optimal, there is way too much movement in the generated g-code that would not
be necessary with optimizations.

After watching my current print on the first two layers I loaded the .stl into PrusaSlicer and cut it down to the bottom 1mm:
[test.zip](https://github.com/prusa3d/PrusaSlicer/files/7868463/test.zip)

With this the result in slicing is the same.

So for the sake of providing better screenshots I rotated the platter anti-clockwise in PrusaSlicer.

The skirt starts at the bottom right in the image, this is fine, minimal travel from the homing point.
But the first hole is generated across where the skirt ends.

[image]

The second hole is also fine:

[image]

The third and fourth hole and the start of the outer perimeter (seam: nearest) is fine as well:

[image]

The first layer infill is almost fine as well

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

The setting for the "top fill pattern" is "monotonic", so the order is for the most part correct.
There are gaps left that are filled at the end, this could be more optimal but I guess can not really due to "monotonic".

So the first layer ends in the bottom left.
The second layer starts with the same hole at the top left as the first layer which is unnecessary travel.

[image]

The rest of the holes and the perimeter are the same as the first layer which is ok after the wrong start of the layer:

[image]

The second layer infill starts on the top left which again is too much travel, it should start at the bottom right:

[image]

The second layer ends at the bottom right:

[image]

The third layer starts at the top left hole again instead of in this case the bottom right:

[image]

So after I went thru all the steps the main issue seems to be that the layer for some reason always starts at the top right hole.
This is not optimal the first time and this is not optimal on all the layer changes.

RudolphRrr, 2022-01-21:
I just noticed that somehow the issue is getting worse with two of the same objects:
[image]

Start is bottom left, why on earth is the first hole to be printed the one that is the farthest away from the end of the skirt?

Apart from travel time that adds up there is annother issue with this.
The smallest ooze on travel can lead to a fine string of filament to be put down on the travel path.
And while this is usually not an issue in the next layers, this is visible on the first layer.

Please fix the way finder.

RudolphRiedel, 2022-03-20:
Re-Tested with 2.4.1 and it behaves the same, the hole that is the farthest away from the starting point is printed first.

This changes a little when I enable Print Settings/Infill/Infill before perimeters:

[image]

But it still starts with the object that is farther away.
And watching the other layers still shows a lot of unneccesary movement. The perimeters for the holes are always put down in the same order, starting from the top right going anti clockwise. Regardless of where the printhead head is before putting down the first hole perimeter, it moves across the whole object while it could just do the hole perimeter at the current position first.

Start of second layer:
[image]

First move on second layer:
[image]

Infill down on second layer:
[image]

Start of perimeters on second layer:
[image]

End of perimeter on the first object:
[image]

Start of infill on the second object:
[image]

This issue does affect print time and print quality as unneccessary moves are performed.

Tupson444, 2026-09-12:
still relevant