prusa3d/PrusaSlicer #7746: Missing ;TYPE comment

Duracell95, 2022-01-14:
**Version**
2.4.0

**Operating system type + version**
Windows 10 64 

**3D printer brand / version + firmware version (if known)**
not relevant

**Behavior**
The ;TYPE comment is not inserted in gcode when the layer only consists of External perimeter

[image]

I expect it to throw the ;TYPE:External perimeter comment regardless if it will change later on or not.

[image]

Might not seem like an issue but i need the comments for scripting outside of PrusaSlicer.

_Is this a new feature request?_
If it's not possible as of now, then yes.

n8bot, 2022-01-14:
I think this is because the last role to be extruded was external perimeter. All the layers of the chimney are external perimeters. If it doesn't change, it's not identified.

lukasmatena, 2022-01-16:
Could you post a 3MF? As far as I know, the comment only appears when the extrusion type changes. If the last thing printed on the layer before was external perimeter, it would not be emitted.

Deuracell, 2022-01-16:
> Could you post a 3MF? As far as I know, the comment only appears when the extrusion type changes. If the last thing printed on the layer before was external perimeter, it would not be emitted.

Hi, yeah that is the case. I managed to rewrite my script in a way this doesn't matter. So alls well, thank you!

closed 2022-01-16 by Deuracell