prusa3d/PrusaSlicer #7750: Need clarification on seams and feature request for making representation of support temporarily invisible

laptop-laptop, 2022-01-15:
### Version
2.4.0

_Use `About->About PrusaSlicer` for release versions_
2.4.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
i3MK3S+

### Behavior
I have Support set to "Everywhere". After pressing 'Slice now', PS generated G-code. Then, I was able to set 'Feature type' in View and check Seams under 'Show Options'. Here is the problem: Some parts of my object are not visible as they are covered by various types of supports (green). So for example, although on one side of a wall does not show any seams, the other side could have seams. However, I cannot tell as the supports cover some parts of the object. Is it possible to add an option to make the supports temporary invisible so we could see if there is any seams in parts covered by Supports?

Another question is if now I temporarily set Supports to None and Slice to see the seams' locations from any angle, will the seams be located at the same places compared with those I had when Supports was set to Everywhere? I guess not but please confirm.

_Is this a new feature request?
Yes

neophyl, 2022-01-15:
If you click a type of feature in the key displayed when sliced it turns the display of the feature off.  So what you are requesting is already possible.

lukasmatena, 2022-01-15:
> Another question is if now I temporarily set Supports to None and Slice to see the seams' locations from any angle, will the seams be located at the same places compared with those I had when Supports was set to Everywhere?

No. The rest is already answered by @neophyl.

closed 2022-01-15 by lukasmatena

laptop-laptop, 2022-01-15:
> If you click a type of feature in the key displayed when sliced it turns the display of the feature off. So what you are requesting is already possible.

Thank you.