prusa3d/PrusaSlicer #7753: Travel through externel perimeter with avoid crossing perimeter enabled

tilation88, 2022-01-15:
[fault with stl file.zip](https://github.com/prusa3d/PrusaSlicer/files/7875113/fault.with.stl.file.zip)

### Version
2.4 stable - linux  (and  also tested on win10, same Version)

### Operating system type + version
linux, same on win10

### 3D printer brand / version + firmware version (if known)
printed on a CR-10s pro

### Behavior
open the attached file
use the config
slice

i think the travel is to close to the external perimeter, what often causes stringing while printing inner radii.

cura in comparison tries to keep distance from the external perimeter when "avoid crossing perimeter" is enabled

all helpfull files are attached. 

sorry for my bad english.

dartrax, 2022-01-16:
Can you please check if what you are seeing is already reported at #5689 ?

lukasmatena, 2022-01-17:
I hope this is fixed in current master and ready for 2.4.1 patch release (see #7699):
[image]

Closing the issue, in case the issue is still there in 2.4.1, let us know and we can reopen it.

closed 2022-01-17 by lukasmatena