prusa3d/PrusaSlicer #7754: 2.4.0 GTK version does not remember window size

jfbauer432, 2022-01-15:
### Version
2.4.0+linux-x64-GTK

PrusaSlicer-2.4.0+linux-x64-GTK3-202112211623.AppImage
or
PrusaSlicer-2.4.0+linux-x64-GTK3-202112211623.tar.bz2

### Operating system type + version
Ubuntu 21.10

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S+ (kit, unmodified)
firmware: 3.10.0-4481

### Behavior
prusaslicer (GTK version) does not remember/use the previous size of the main window.  It does remember window position (and if it was fullscreen), just not size.

1) run prusaslicer (no arguments)
2) make sure it is not fullscreen and note window size
3) exit prusaslicer
4) restart prusaslicer and note window size not same (according to xwininfo size is 890 wide 598 high)

Expected: that prusaslicer would use previous window size

If you do this with the non-GTK version, it works as expected.

#### Project File (.3MF) where problem occurs

N/A

KlausHans, 2022-03-13:
Very annoying issue, unfortunately not solved in 2.4.1 :(

lukasmatena, 2022-03-14:
Thanks, we will look into it one day.

HappyCodingRobot, 2022-05-14:
Can confirm this issue on 2.4.2

KlausHans, 2023-04-29:
Over one year later still annoying :/
Maybe in 2.6, @lukasmatena? Would be great.

lukasmatena, 2023-05-02:
@KlausHans Actually, it is already fine in 2.6.0-alpha6. Closing.

closed 2023-05-02 by lukasmatena