prusa3d/PrusaSlicer #7755: Slow to open/quit on MacOS

tomotvos, 2022-01-15:
### Version
PrusaSlicer-2.4.0+x64-202112211607

### Operating system type + version
MacOS 12.1 (Monterey)

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior
* Launching PrusaSlicer seems to take a very long time, as does quitting the app

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
n/a

FidelCapo, 2022-01-17:
Unfortunately, I cannot reproduce this behavior. Any special OS settings? External monitors connected? Is it slow even in 2.3.3 or only in 2.4.0?

tomotvos, 2022-01-17:
No special OS settings that I can think of, and definitely no external monitors. I can try and re-download 2.3.3, but my guess is that this is new with 2.4 since I did not notice anything before, until I did.

tomotvos, 2022-01-17:
Hmm, 2.3.3 is pretty responsive, and so now is 2.4. I'll keep an eye on this and see if I can provide any other clues. Can you keep this open a bit and if I don't update in the next few days it can be closed?

FidelCapo, 2022-01-18:
Yes, no problem. Let me know if you have any more info. Thanks

closed 2026-09-29 by prusaslicer-bot[bot]