prusa3d/PrusaSlicer #7757: PrusaSlicer Shift & TAB shortcut not working

Jonny98731, 2022-01-15:
### Version
2.4.0

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa i3M3S+, Version not relevant for my report

### Behavior
I have a standard European Keyboard and the shortcut for hiding the right side menu does only work with the right shift key... Please implement it so that it works for both keys, right and left bcause i usually use the left shift key :)

_Is this a new feature request?_ 
Kind of

lukasmatena, 2022-01-17:
It should already work for both shifts. Is there anything special about your keyboard settings? Don't you use shift for some accessibility function or somethings like that? Is PrusaSlicer 2.3.3 affected as well? Thanks.

Jonny98731, 2022-01-17:
> It should already work for both shifts. Is there anything special about your keyboard settings? Don't you use shift for some accessibility function or somethings like that? Is PrusaSlicer 2.3.3 affected as well? Thanks.

I just tried out a few things. I figured out that my issue is on both sides: It works on the left side if i keep TAB pressed, then press Shift L and then realease TAB again. Somehow if i press shift first, it does not work. Also doesn't work on the other side.

But this effect occurs only sometimes and i don't know why. I also don't know how it is coded but my guess would be that it's about for which input of this two keys is the code waiting for first if that makes sense. I'm no programmer so please forgive me :)

Also it's not a big deal so don't waste time on it if you don't have time for it right now ;) Your doing a great job with PrusaSlicer.

lukasmatena, 2022-01-18:
@Jonny98731 Thanks for the investigation. Could you please try this build: https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.4.1-alpha0+19-lm-keymodifiers-win64-g3c619bf8d-202201171528.zip ?
I doubt it will help to be honest, but there is a slight change in how it is handled. And since we cannot reproduce it ourselves, it would help if you tried it out.

Jonny98731, 2022-01-19:
Tried the build, made no difference.
I have a Microsoft Internet Keyboard Pro from 2001. Never had any issues in other programs, but maybe this makes a difference. Anyways - since i know now that i need to press TAB first, it makes the shortcut even useful to me ;)

ulab, 2023-03-12:
I have noticed a similar behaviour on Mac. 

Once I have clicked anywhere in the sidebar shift+tab will move the focus between the different form fields / windows - as I would expect. In that case shift+tab will obviously not work anymore to close the sidebar.

Perhaps there should be a different shortcut?

closed 2026-09-29 by prusaslicer-bot[bot]