prusa3d/PrusaSlicer #7758: Slicer does not save settings and changes

daylie-01, 2022-01-15:
Hello everyone..I have a "small" problem with Prusa 2.4 (version 2.4.0+win64)..
If I want to make an extruder change for an object in print and use the slider to the right of the object to set the height when it should make the change, the setting is gone again before slicing.. With version 2.3 it works great..Hab I've installed the new version twice now, but it didn't help. I can't enter my own G codes either, they are deleted immediately (e.g. with a TempTower). As I said, everything works fine in V 2.3..
Printer is a Geeetech A10T.. Did I forget some setting in the new version?
I've attached a video that might help with troubleshooting.
Thanks in advance..
p.s. Notebook Dell G5 with Win 11..

https://user-images.githubusercontent.com/97799350/149631585-9c7df8b3-6edc-4429-9620-8495bd737c38.mp4

lukasmatena, 2022-01-16:
Post a 3MF.

daylie-01, 2022-01-16:
Hi there,
I can only post something like this as a zip file, 3mf is not supported here.
[Chicken_with_eyebrows.zip](https://github.com/prusa3d/PrusaSlicer/files/7876536/Chicken_with_eyebrows.zip)

lukasmatena, 2022-01-17:
Thanks for the report, you are right. There is some problem in handling the "Z offset" parameter from Printer Settings. We'll look into it.

daylie-01, 2022-01-22:
Hi there,
Is there now a solution to the problem?

YuSanka, 2022-01-22:
Yes, this issue is fixed with commit https://github.com/prusa3d/PrusaSlicer/commit/76ce53f09563de6f89460efb0300a675193e0823 and will be a part of upcoming release.

daylie-01, 2022-01-22:
Super thank you.
The support is absolutely great.

daylie-01, 2022-01-24:
Hi there,
do you already have an appointment for the update?

daylie-01, 2022-02-05:
Hi there,
I now have the 2.4.1 beta running and everything works as it should. Thank you

closed 2022-02-05 by daylie-01