prusa3d/PrusaSlicer #7759: Slicer 2.4.0 adds extra vertical line to a part of a straight wall

udeboer, 2022-01-15:
I have an enclosure where the gcode generation adds extra vertical lines on otherwise straight walls. And only on part of that wall. See example screenshots and 3mf file. Because it is appears on a flat vertical surface I cannot see why it would occur.

### Version Slicer
2.4.0 linux x64 GTK3
Linux is Ubuntu 20.04 with current HWE kernel

Printer is MK3S
Included screenshots from model en generated gcode. And the 3mf file in zip format

Model
[image]
Gcode shows extra lines
[image]
[enclosure.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7875493/enclosure.3mf.zip)

udeboer, 2022-01-15:
I found at the reason. It is caused by the seam position setting on "nearest". Before I never noticed the seam so much in my prints. I did found out that some retraction settings where changed in the newer version of the defaults. So maybe this has something to do with it. I was still using the old 1,4 mm setting on the Prusa PETG Filament overwrite settings page.

bubnikv, 2022-06-29:
PrusaSlicer 2.5.0-alpha2 will align the seam line better.

closed 2022-06-29 by bubnikv