prusa3d/PrusaSlicer #7761: Custom Start & End G-Code does not convert placeholders to proper units (from mm/s to mm/min)

linyongzheng, 2022-01-15:
### Version
2.4.0+win64

### Operating system type + version
Windows 10 Pro 21H1

### 3D printer brand / version + firmware version (if known)
Any (reproduced on multiple printers profiles)

### Behavior
Custom start & end G-code does not convert placeholder feed rates from mm/s to mm/min, resulting in such placeholders traveling only 1/60th the intended feed

Insert travel place holder ([travel_speed] or (travel_speed_z]) into custom start/end g-code
example: G1 Z0.2 F[travel_speed_z] (where travel_speed_z is set to 12)
[image]

* _Expected Results_
exported g-code of G1 Z0.2 F720

* _Actual Results_
exports g-code G1 Z0.2 F12
[image]

#### Project File (.3MF) where problem occurs
[RJ45-cover.zip](https://github.com/prusa3d/PrusaSlicer/files/7875873/RJ45-cover.zip)

rtyr, 2022-01-15:
Expected behavior, use `F{travel_speed_z*60}` instead.

https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-placeholders-(a-copy-of-the-mauk.cc-page)

linyongzheng, 2022-01-16:
Thanks. I guess I'll add a feature request to have the slicer translate to the appropriate units so that other newbies like me won't make this mistake.

closed 2022-01-16 by linyongzheng