prusa3d/PrusaSlicer #7762: 2 issues I came across: adds unwanted perimeters to modifier box // incorrect slicing

Michl59, 2022-01-15:
### Version
2.4.0 (win and mac)

### Operating system type + version
Windows 10 Pro and Mac OS Catalina

### 3D printer brand / version + firmware version (if known)
not yet printed due to issue

### Behavior
1st issue:
when adding modifier box to reenforce the **existing** perimeters within the box, Slicer **adds** perimeters inside the modifier box.

(if this behaviour is a feature instead of a bug, it would be nice to have an option to enforce existing perimeters without adding some. Useful for wholes made for thread inserts where you need extra stability but other perimeters are fine in a lower number)

2nd issue:
when slicing a print that has a 0.5mm wall thickness and trying to slice it with a 0.5mm extrusion width the print is "scattered" and at lower widths the line gets printed twice(!)
issue with round object, also with non round shaped objects.
My expectation was that an object with a wall thickness of 0.5mm will be printed with a single pass at 0.5mm extrusion width.
As widths of 0.49 and lower get a second pass I wonder if this is intended. 0.25 works fine in Slicer as it makes two passes to generate the 0.5mm perimeter. However, this seems challenging with a 0.4mm nozzle

I've attached various examples and pictures for the issues above, issues is present on both win and mac os

Thanks for looking at this :)

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Jebtrix, 2022-01-24:
> 1st issue:
> when adding modifier box to reenforce the existing perimeters within the box, Slicer adds perimeters inside the modifier box.

That would be nice though way beyond trivial. Luckily adding a solid infill modifier around the holes is stronger than adding perimeters. 

[image]

If you really wanted more perimeters it would take more CAD work. A lot more work for less strength.
[image]

> 2nd issue:
when slicing a print that has a 0.5mm wall thickness and trying to slice it with a 0.5mm extrusion width the print is "scattered" and at lower widths the line gets printed twice(!)

Scattered is usually a curve issue, the math just needs some fudge tolerance room sometimes. Here is high poly 0.5mm thick pipe like your project I exported with maximum precision. If you actually measure it, it is not 0.5mm _everywhere_. Between the outer edges its actually below. This is why you get scatter, 0.5mm extrusion doesn't mathematically fit. 

[image]

[image]

The double loop when it should do one is definitely an issue. If you slightly tweak extrusion wider it will even double loop in a segmented backtrack manner. Enabling thin wall detection option does work with this specific object without its usual side effects.

For the devs ([Single_Perimeter_Issue.zip](https://github.com/prusa3d/PrusaSlicer/files/7922690/Single_Perimeter_Issue.zip)):

[image]

Michl59, 2022-01-26:
Hi Jebtrix!

Thank you for the response, very helpful! :)

regarding the additional perimeters:
The additional perimeters were there mainly for the reason that I will insert brass threads (the Rutex/CNC Kitchen ones) and those require some "meat" around the holw in order to work properly. So this is me thinking adding perimeters with a modifier is easier/faster than everything else, turns out I misinterpreted the functionality :)

Also I opted for more perimeters instead of more infill as I didn't need maximum strenthg and Stefan (CNC Kitchen) found out that more perimeters make the part stronger vs just more infill. So all in all I thought the "quick" solution was to add more perimeters, which turned out to be wrong

2nd issue:
Thanks for the work explaining this, I was in need of a part that adapted a 5mm axle to a 6mm axle holw in a gear for a stepper motor. This is probalby more of a use case for the full control G-Code generator based on Excel, where I should tell the printer to print the part with one motion and an extrusion width on 0.5mm

Curioius it would do the double loop and als segmented backtracking... Found this on other slices as well where it should move in one go but doesn't. Unfotunately I didn't keep those, would it help if I post these here in case I come across this again?

Cheers!

Michael

Jebtrix, 2022-01-26:
The results I have for parts in service is more perimeters isn't stronger. But like anything there may be exceptions depending on shape and purpose along with all the other almost infinite variables.

Unless a dev says the single loop/trace extrusion behavior is normal it can't hurt to post a couple example project files for them to investigate.

bubnikv, 2022-01-28:
The 2nd issue will improve if you enable "detect thin walls"

Michl59, 2022-01-28:
> The results I have for parts in service is more perimeters isn't stronger. But like anything there may be exceptions depending on shape and purpose along with all the other almost infinite variables.

To be honest I'm not very experienced yet, just got my printer last week ^^
Since ordering it after Christmas, I watched and read a ton on 3D Printing. Stefan over at CNC Kitchen made some videos trying different parameters in 3d printing and had a very scientific approach to the tests. There he found that more perimeters have stronger properties compared to more infill. So that is where my approach is from.

https://www.youtube.com/c/CNCKitchen

> Unless a dev says the 1 perimeter extrusion behavior is normal it can't hurt to post a couple example project files for them to investigate.
Will do!

closed 2026-09-29 by prusaslicer-bot[bot]