prusa3d/PrusaSlicer #7763: Feature request: add unit conversions for placeholders in custom g-code (convert speed placeholders to mm/min)

linyongzheng, 2022-01-16:
### Version
2.4.0-win64

### Operating system type + version
Windows 10 Pro 21H1

### 3D printer brand / version + firmware version (if known)
Any

### Behavior
* _Describe the problem_
Placeholders are just the GUI values in custom G-Code section instead of the appropriate units.
* _Steps needed to reproduce the problem_
Place code "G1 X-20 F[travel_speed]" into start G-code section of custom g-code where travel_speed is 250 [mm/s]
* _Expected Results_
Exported g-code should have be "G1 X-20 F15000" because feed should have been converted to mm/min
* _Actual Results_
Export g-code is "G1 X-20 F250" because the units aren't converted

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
[RJ45-cover.zip](https://github.com/prusa3d/PrusaSlicer/files/7876159/RJ45-cover.zip)

Alpacad, 2022-01-16:
@linyongzheng You can use a **macro** for that
This should do the trick

G1 X-20 F{[travel_speed] * 60}

Didn't try it tho
As for the feature, maybe add a **method** that can be attached to the placeholder like
F{Convert.mmMin[travel_speed]}
but this looks way too complex than using the actual formula

lukasmatena, 2022-01-17:
I think that the responsibility for converting the value to the ~~correct~~ **desired** value should lie on the user. Different firmware flavours for example use different units, the placeholder parser has too little context to get everything right. In my opinion, it should just get the config value and not try to be smart, which would only be confusing. The user can easily convert the value himself in the custom gcode (as illustrated by @Alpacad and described in more detail [here](https://help.prusa3d.com/en/article/macros_1775)).

Closing. If anyone thinks this should be reconsidered, we'll happily hear the reasons.

closed 2022-01-17 by lukasmatena