prusa3d/PrusaSlicer #7765: Is it possible to change the order of printing support and the actual object within the same layer?

laptop-laptop, 2022-01-16:
### Version
2.4.0

_Use `About->About PrusaSlicer` for release versions_

2.4.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa i3MK3S+

### Behavior
* _Describe the problem_

So far within the first layer, Prusa Slicer prints the support surrounding the object first and then outline of the actual object followed by infill. Is it possible to change the order so that it prints the outline of the actual object first, then infill and lastly the support?

_Is this a new feature request?_

Yes if this feature is not available

lukasmatena, 2022-01-17:
It is currently not possible What is the reasoning for the feature request? Thanks.

n8bot, 2022-01-17:
There is a distinct difference in the visual quality of the external print perimeters depending on the order the extrusion roles are printed in. 

The support extrusions squish against the perimeters differently depending on which line is printed first.

laptop-laptop, 2022-01-18:
> It is currently not possible. What is the reasoning for the feature request? Thanks.

Thanks for confirmation

closed 2022-01-18 by laptop-laptop