prusa3d/PrusaSlicer #7766: Bridging Not Being Detected

djngnear, 2022-01-17:
### Version
2.4.0+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Prusa Mk3s+, v3.10

### Behavior
* _Describe the problem_
Bridging is not being detected when it clearly should be.
* _Steps needed to reproduce the problem_
* 
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
Blue bridges should show up in between the structures
* _Actual Results_
Normal perimeters show up
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
  * 
[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
https://drive.google.com/file/d/122NS-7U_RGSViI6gMcvad6OTb25uBLKP/view?usp=sharing
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

djngnear, 2022-01-17:
[image]

lukasmatena, 2022-01-17:
Thanks for the report. You are right, the detection fails in this case. It works roughly by offsetting the slice below outward slightly and than considering everything not inside the result as a bridge. What is probably happening here is that the below slice already contains a very narrow bridge, which satisfies the criteria despite the fact that no extrusion is actually laid on it. We'll try to improve on this in the future.

combs, 2022-01-19:
I'm seeing what might be a similar problem with the Mk3S `fan-shroud.stl`—the bridge is not detected and it prints poorly.

djngnear, 2022-01-19:
> I'm seeing what might be a similar problem with the Mk3S `fan-shroud.stl`—the bridge is not detected and it prints poorly.
> 
> notice lack of overhang/bridge detection on the big central bridge:
> 
> [image]
> 
> prints like this:
> 
> [image]
> 
> update--it looks like there's a [newer version of this file](https://github.com/prusa3d/Original-Prusa-i3/commits/MK3S/Printed-Parts/STL/fan-shroud.stl) that includes a manual support. So that will fix my short-term problem. But interesting that the changes to no longer detect "knife edge" bridges also affected Prusa parts.

Yes I’ve noticed it happening on really narrow extrusions. To get around this I’ve been messing with the settings in SuperSlicer and have gotten the correct results I’m looking for. Hopefully the next release of PS will fix it.

Drak0sha13, 2023-09-13:
The bridge is not detected. The line just hangs in the air.

[STL](https://cloud.uni-3d.ru/s/GkfasHfKFzX5toy/download?path=%2F&files=%D0%A1%D0%BE%D0%BF%D0%BB%D0%BE%20%D0%BE%D0%B1%D0%B4%D1%83%D0%B2%D0%B0%20%D0%BC%D0%BE%D0%B4%D0%B5%D0%BB%D0%B8.STL)
[image]

PS. The classic perimeter generator did not close the mounting holes.

closed 2026-09-29 by prusaslicer-bot[bot]