prusa3d/PrusaSlicer #7768: Feature request: Aligned Honeycomb infill pattern

YvonnevH, 2022-01-17:
### Feature request
I would like to request an aligned honeycomb infill pattern. The current honeycomb infill changes orientation every layer, so that each time 2 lines are printed on top of 1 line (see image). For cleanliness reasons when infill pattern is a visible part of the model, I would like to have an aligned honeycomb pattern as well, just as the rectilinear and aligned rectilinear patterns. 

Based on the code, it seems that adding a new infill pattern 'Aligned Honeycomb', could be done by duplicating the regular honeycomb pattern, and adjusting only the following line. 

`float _layer_angle(size_t idx) const override { return float(M_PI/3.) * (idx % 3); }`
to 
`float _layer_angle(size_t idx) const override { return 0.f; }`

[image]

neophyl, 2022-01-17:
You really need this feature request https://github.com/prusa3d/PrusaSlicer/issues/2293
As that would allow you to set the 'step' angle of the infill which would be far more flexible than adding another infill type.  

Super Slicer already has this function and if you set a 120 degree step then the honeycomb in effect prints the same all the way up (I just tried it).

gringer, 2022-07-06:
PR #6434 adjusts the current 3D honeycomb pattern to an aligned patern, fixing the alternating line issue of the current implementation. It does switch orientation every half-cell (similar to what gyroid does), but that's a "natural" switching point which makes the structure a little bit more stable.

closed 2026-09-29 by prusaslicer-bot[bot]