prusa3d/PrusaSlicer #7770: Generic Slab + Ironing = Ironing on Solid Infill(?!) and EVERY Flat layer (issue caused quick 20 minute print to take 1h22m!!)

todsandberg, 2022-01-17:
### Version
_Version of PrusaSlicer used goes here_
2.4.0

### Operating system type + version
Win 10 Pro Build 19043.1466

### 3D printer brand / version + firmware version (if known)
Prusa MK3s+ (firmware N/A. Issue is with slicer)

### Behavior
I had a part where I only wanted ironing on a single flat surface, so I added a slab with the intention that adding ironing would iron the top surface, instead it Ironed EVERY FLAT LAYER of the slab (not just the top layer).

I expected this

> **Ironing**
> Layer
> Layer
> Layer
> Layer
> Layer Infill
> Layer Infill
> Layer Infill
> Layer Infill
> Layer
> Layer
> Layer (bottom)

**This is what happened:**

> **Ironing**  << Only layer that should be ironed.
> Layer
> **Ironing**  << Shouldn't happen
> Layer
> **Ironing**  << Shouldn't happen
> Layer
> **Ironing**  << Shouldn't happen
> Layer
> Layer Infill
> Layer Infill
> Layer Infill
> Layer Infill
> **Ironing**  << Shouldn't happen
> Layer
> Layer
> Layer (bottom)

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_  
**_Could not find this option. So uploading in native 2.4.0 version._**

todsandberg, 2022-01-17:
This issue caused a quick 20 minute print to take 1h22m due to all the unneeded ironing!!

todsandberg, 2022-01-17:
[ISSUE Dishwasher_slide (FAIL ironing applied to slab causes each layer to be ironed!!).zip](https://github.com/prusa3d/PrusaSlicer/files/7885132/ISSUE.Dishwasher_slide.FAIL.ironing.applied.to.slab.causes.each.layer.to.be.ironed.zip)

djngnear, 2022-01-18:
I got it to work by also adding the ironing type modifier for the generic slab and selected all top surfaces.  Maybe by default it chose all solid surfaces to be ironed instead.

[image]
.

todsandberg, 2022-01-18:
Thanks djngnear.

I meant to upload the photos earlier . Here are shots showing each layer with ironing.

**OVERVIEW IMAGE
[image]

LAYER 13
[image]

LAYER 14
[image]

LAYER 15
[image]

LAYER 16
[image]

LAYER 17
[image]**

FidelCapo, 2022-01-18:
For the modifier the "Enable ironing" setting is added but the "Ironing Type" is not set. In Print settings -> Infill -> Ironing -> Ironing Type you have set "All solid surface" and this setting is used because it is not defined for the modifier. If I change this to "All top surfaces" everything is ok in my opinion.

closed 2026-09-29 by prusaslicer-bot[bot]