prusa3d/PrusaSlicer #7771: Unable to correctly slice the number 4 in text

helo-head, 2022-01-18:
### Version 
2.4.0

### Operating system type + version. 
MacOS Monterey 12.1

### 3D printer brand / version + firmware version (if known) 
Prusa MK3S

### Behavior
Prusa slicer does not seem to be able to fully form the number 4 no matter the font or size exported. It appears to drop the bottom of the number. Refer to screen shot file

#### Project File (.3MF) where problem occurs

See attached

[Fillet Tool v4.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7885906/Fillet.Tool.v4.3mf.zip)
[image]

fe60, 2022-01-18:
Looks strange indeed and I'm able to reproduce the issue.
In my setup (0.4mm nozzle) I get a better result reducing the extrusion width for the numerals to 0.28mm.
(Height range modifier=>Range 3.0-4.0mm (0.2mm layer height), Add Settings=>Extrusion Width=>External perimeter: 0.28)

[image]

helo-head, 2022-01-22:
Your recommendations helped once I figured out that something odd is happening when changing printers. I found the only way to get your setting to work was to change back to original prusa MK3S printer (.4 nozzle), set the external perimeter and then go back to my modified MK3s  setting (.5 Nozzle). Unless I did it this way your setting had no effect. Not sure why but at the end of the day, rule 25 says “Never Argue with Success”.

closed 2026-09-29 by prusaslicer-bot[bot]