prusa3d/PrusaSlicer #7773: Preview tool head location incorrect

0x3333, 2022-01-18:
### Version
2.4.0+x64

### Operating system type + version
Mac OS 10.15.7

### 3D printer brand / version + firmware version (if known)
CR 20 Pro - Marlin 2.0

### Behavior
If I let my start gcode, the preview in Prusa Slicer thinks that the tool head is in a position it is not. I'm using G90 in the start gcode and it is summing the current position(start gcode) with the position in the sliced output.
It was working great on version 2.3.3.

The tool position is the sum from the last position in the startup code, and the absolute position in the generated gcode(last image).

Last Y position in my start gcode: 80
First Y position in the generated gcode: 65.828
Position in the preview window: 145.828

```gcode
; generated by PrusaSlicer 2.4.0+x64 on 2022-01-18 at 14:46:40 UTC

; 

; external perimeters extrusion width = 0.40mm
; perimeters extrusion width = 0.40mm
; infill extrusion width = 0.40mm
; solid infill extrusion width = 0.40mm
; top infill extrusion width = 0.40mm

M73 P0 R390
M201 X500 Y500 Z100 E1000 ; sets maximum accelerations, mm/sec^2
M203 X150 Y150 Z12 E100 ; sets maximum feedrates, mm/sec
M204 P500 R500 T1500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X8.00 Y8.00 Z2.00 E10.00 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
;TYPE:Custom
M104 S235     ; Set extruder temp
M140 S100 ; Set bed temp

M117 Waiting temperature...
M190 S100 ; wait for bed temp
M109 S235     ; wait for extruder temp

G90 ; use absolute coordinates
M83 ; extruder relative mode

G28                     ; Home all axes
G0 Z 7 F50000           ;
G0 X 239 Y 0 F5000      ; Go to nozzle's purge position
G0 Z0.15 F5000

M117 Purging nozzle...
G92 E 0                  ; Reset Extruder
G1 F 200 E 15            ; Purge nozzle
G0 X 228 Z 0.3 F 5000    ; Go to nozzle's purge line position
G1 Y80 E6 F700           ; Extrude 6mm of filament in a 7cm line
G92 E 0                  ; Reset Extruder
G0 X 225 Z 0.1 F2000     ; Quickly wipe away from the filament line

M117 Printing...
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
; Filament gcode
M107
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
; BEFORE_LAYER_CHANGE 0 @ 0.2mm
G1 Z.2 F4200
; AFTER_LAYER_CHANGE 0 @ 0.2mm
G1 E-5 F6000
G1 X80.031 Y65.828 F4200
G1 E5
;TYPE:Skirt/Brim
;WIDTH:0.4
G1 F1500
G1 X82.429 Y62.982 E.11048
...
```

[image]
[image]
[image]

FidelCapo, 2022-01-18:
Can you please attach the zipped 3mf file? File -> Save Project As... Thank you.

0x3333, 2022-01-18:
It happens with all models. But here it is. Its a Voron Printer part.

[stealthburner_toolhead_(dragon)-front-beta2.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7892033/stealthburner_toolhead_.dragon.-front-beta2.3mf.zip)

0x3333, 2022-01-18:
Just to clarify, the output itself is correct, I can print the part no problem, it is a Preview problem only.

enricoturri1966, 2022-01-19:
0x3333
Thank you for your report.
We found a bug in the gcode parser which was introduced after 2.3.3 release.
I have committed the fix which will be released soon into 2.4.1 release.
As workaround, you can have your 3mf working in 2.4.0 by removing all the whitespaces in between the coordinate axis digit and its value from all your custom gcode.
For example line:
G92 E 0
should be written as:
G92 E0

0x3333, 2022-01-19:
OMG, I just changed my gcode adding those spaces as it made it clearer to read. Changed and it worked. Thanks for the awesome work!

closed 2022-01-19 by 0x3333