prusa3d/PrusaSlicer #7774: MKS.WIFI support needs handling long and non-ascii file names.

RandoMan70, 2022-01-18:
### Version
_v2.4.0_

### Operating system type + version
Any operating system

### 3D printer brand / version + firmware version (if known)
mks.wifi family

### Behavior
mks.wifi does not handle long names and localized strings, that leads to errors and sometime printer resets.

bubnikv, 2022-01-18:
I think the MKS.WIFI pull request was from you, right?

To get rid of the non-latin7 characters, you need to call one of

```
// If possible, remove accents from accented latin characters.
// This function is useful for generating file names to be processed by legacy firmwares.
extern std::string 	fold_utf8_to_ascii(const std::string &src);

// Convert the input UNICODE character to a string of maximum 4 output ASCII characters.
// Return the end of the string written to the output.
// The output buffer must be at least 4 characters long.
extern wchar_t* fold_to_ascii(wchar_t c, wchar_t *out);
```

RandoMan70, 2022-01-18:
@bubnikv , sure, assign it to me is possible. Thank your for advice. I'll also investigate maximum length of filename for mks.wifi.

closed 2026-09-29 by prusaslicer-bot[bot]