prusa3d/PrusaSlicer #7777: Regression issue slicing output 2.3.3 -> 2.4.0

raven-worx, 2022-01-19:
### Version
PrusaSlicer 2.4.0

### Operating system type + version
Windows 11

### 3D printer brand / version + firmware version (if known)
Prusa Mini+ - FW 4.3.4
Bootloader 1.1.0
Buddy Board 1.0.5

### Behavior

I several times tried to print a simple gear/cog, sliced with PrusaSlicer 2.4.0 (GENERIC PLA, 0.1mm DETAIL - just reduced all the extrusion speeds of infills and perimeters).
I always ended up with the noozle shifting already printed layers a side. Just like the Z-axis value isn't increased anymore upon a specific point. After several tries what might be the issue i simply tried to slice it with the previous version of PrusaSlicer (2.3.3) and it printed successful on the first try.

#### Project File (.3MF) where problem occurs
STL, project files, settings and gcode results of 2.3.3 (ok) and 2.4.0 (nok)
[issue_files.zip](https://github.com/prusa3d/PrusaSlicer/files/7897267/issue_files.zip)

rtyr, 2022-01-19:
I don't see any issue in the gcode and I printed it on the MINI without problem.

The issue is likely not related to PrusaSlicer. Probably some mechanical issue or corrupted filesystem on the flashdrive.

Please contact our tech support at www.prusa3d.com for further troubleshooting.

[image]

raven-worx, 2022-01-20:
strange, i checked again by printing the nok gcode file and ended up with the same issue. The last print was the successful ok gcode print.
I am printing with OctoPrint via serial connection.

rtyr, 2022-01-20:
It may be some Octoprint or Octoprint/FW issue, you may try to print it directly from USB.

raven-worx, 2022-01-21:
It printed fine from usb stick.
I will check my octoprint setup.
Thank you!

closed 2022-01-21 by raven-worx