prusa3d/PrusaSlicer #7778: Download of wxWidgets dependency fails

margre8, 2022-01-19:
### Version
2.4.0

### Operating system type + version
Ubuntu 20.04

### Behavior
When I tried to build `PrusaSlicer version 2.4.0` the cmake build dependencies script failed on download of wxWidgets. In the terminal log I can see the expected sha256 hash is different than actual hash of `wxWidgets/v3.1.4-patched.zip`.

tamasmeszaros, 2022-01-19:
Hi @margre8  Delete the old archive. This issue will be fixed soon.

margre8, 2022-01-19:
@tamasmeszaros Thank you for answer. Do you mean I have to delete `wxWidgets/v3.1.4-patched.zip.` or entire PrusaSlicer source code directory?

tamasmeszaros, 2022-01-19:
Delete just `wxWidgets/v3.1.4-patched.zip` the zip will be downloaded again and then it should work.

margre8, 2022-01-19:
@tamasmeszaros This archive is automatically deleted after failure of SHA check. I can't delete it manually.

tamasmeszaros, 2022-01-21:
Yes cmake normally deletes it automatically, so at the next build attempt it succeeds. I've changed the URL recently to point to an exact commit of wxWidgets to avoid this issue.

tamasmeszaros, 2022-01-21:
If you have your build ready, please let us know, then we can close this issue.

margre8, 2022-01-21:
After replacing hash in `deps/wxWidgets/wxWidgets.cmake` build of dependencies was successful. Thank you for your help.

closed 2022-01-21 by lukasmatena